AravisProject/aravis #224: [Question] How to handle error ?

wuyuanyi135, 2019-01-15:
Hi, 
I wonder how to catch the error for this library. For example, if I `arv_camera_new` twice for the same camera, it prints to console and return NULL that I can check. What if I try to acquire image after the camera is unplugged. Is there any defined way to handle it?

Thank you

EmmanuelP, 2019-01-15:
You may either want to check the return value of arv_device_get_status, or connect to the 'control-lost' signal.

closed 2019-01-15 by EmmanuelP

wuyuanyi135, 2019-01-16:
@EmmanuelP I checked the source code and tested in my case. It seems we can only check the device status after some operation and then we can know that the device is already offline. Is there any way to listen to/poll the device connection status before performing any operation?

EmmanuelP, 2019-01-17:
There is no such API. Either you connect to the control-lost signal, or you try to read a value from the camera (a value that is not cached), and then check the status.