AravisProject/aravis #225: The arv buffer status has been set to ARV_BUFFER_STATUS_CLEARED

duchy, 2019-01-19:
A Basler GigeV Camera has been used to do over night test. 
It will failed with a wrong buffer status: "ARV_BUFFER_STATUS_CLEARED".
But I found no where the buffer will be cleared.
Does anyone know in which case the arv buffer will be cleared?

Below is the debug logs including the bad frame 32688 which is the one cleared.

[GvStream::check_frame_completion] Completed frame 32687
[GvStream::close_frame] Close frame 32687
[GvStream::find_frame_data] Start frame 32688
[PixelFormat::to_gst_caps_string] 0x01080009 -> video/x-bayer, format=(string)rggb
packet_type  =       ok (0x0000)
content_type =   leader (0x0001)
payload_type = image
pixel format = video/x-bayer, format=(string)rggb
width        = 1278
height       = 1024
x_offset     = 0
y_offset     = 0

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)
[... 413 lines cut ...]
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =    block (0x0003)

packet_type  =       ok (0x0000)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Completed frame 32688
[GvStream::close_frame] Close frame 32688

closed 2019-01-22 by duchy

EmmanuelP, 2019-01-22:
Hi,

I did not have time to look at your report. How did you fix your issue ?

duchy, 2019-01-30:
Hi
In our test, when trying to stop, an empty buffer will be pushed into the buffer queue to notify the consumers waiting on the queue. It is caused by our modifications.