AravisProject/aravis #227: Baumer VLXT Support - New GVSP Version?

hendrikschilling, 2019-02-04:
Hello,

Camera: Baumer VLXT-123M.I (10GigE)
Aravis version: Recent git - 183af87

The camera is recognized correctly and I poked it (with arv-tool) until it started to send data, however the GVSP packages were not parsed.
I looked at the network stream with wireshark, and it seems that the GVSP header is larger. I could identify the relevant fields and have modified aravis to parse these new headers here: https://github.com/hendrikschilling/aravis/tree/vlxt-workaround
I would like to add proper support, but I need help as I don't know anything about the GigE Vision protocol.
Also there are several issues with this hack:
- changes are hardcoded as I don't know how to recognize the new variant, hence all other cameras will break
- resend does not work. Camera answers with short cryptic packets

Attached is the network trace I used.
[cap_test.zip](https://github.com/AravisProject/aravis/files/2827307/cap_test.zip)

EmmanuelP, 2019-02-05:
Looking at your modifications, it looks like aravis is missing support for extended IDs, indicated by GVSP_EXTENDED_ID_BIT flag in wireshark dissector:

https://github.com/AravisProject/aravis/blob/39d58e1295fb3505809eeac1b371f9da946a9d08/tools/wireshark-dissectors/packet-gvsp.c#L1735

hendrikschilling, 2019-02-27:
Okay so what would need to be done to implement support? I believe detection at the packet level does not work. Also is there some free documentation GigE Vision?
I will try to implement this.

EmmanuelP, 2019-02-28:
Hi Hendrik,

GigEVision specification documents are free of charge, but can not be used for aravis development.

A good source of informations is the wireshark dissectors. They are licensed under a GPL licence, and,  even if aravis is LGPL, I guess it is ok to read them for informations, as long as we don't cut'n paste code from them.

I would really appreciate if you can work on this support.

Cheers,

    Emmanuel.

hendrikschilling, 2019-02-28:
Ok, I will try.
I just wondered if you had written any documentation on how GigE Vision works (and how this is implemented in aravis).
I'll look at the dissectors and write back when I have a plan how to implement this. Do you know if the extended id feature is dynamic or fixed (per camera)?
Best Hendrik

EmmanuelP, 2019-02-28:
> I just wondered if you had written any documentation on how GigE Vision works (and how this is implemented in aravis).

There is bit of documentation here: https://aravisproject.github.io/docs/aravis-0.6/aravis-gv.html , but nothing about the implementation, apart the incomplete API documentation.

EmmanuelP, 2019-02-28:
> I'll look at the dissectors and write back when I have a plan how to implement this. Do you know if the extended id feature is dynamic or fixed (per camera)?

It looks like it is dynamic, a bit is set in the stream packet headers saying if extended ids are in use or not.

EmmanuelP, 2019-04-11:
Did you have a chance to work on this ?

hendrikschilling, 2020-02-10:
A bit delayed, but I now have a working prototype which uses extendend ids dynamically depending on the received packages:
https://github.com/hendrikschilling/aravis/tree/vlxt-workaround

It's still a bit hacky, but works reliably including resend (there could be an issue with large blockids - I have to rig up a test for that somehow...).
I do have one question though:
GVSP packets are UDP packages where the payload consist of a header a leader and the actual payload - I have modified the respective structures to support the changed (and larger) leader. What I don't understand is how the memcpy in argvstream.c/_process_data_block which copies from packet->data. How is the leader not copied - which should be part of packet->data?
Cheers
Hendrik

EmmanuelP, 2020-08-04:
Hi Hendrick,

Based on our work and the wireshark gvsp dissector, I have created a branch with extended ids support.

https://github.com/AravisProject/aravis/tree/wip/extended-ids

Could you give it a go ?

EmmanuelP, 2020-08-13:
I have merged the branch into master. Thanks for your work.

closed 2020-08-13 by EmmanuelP

abrock, 2020-08-17:
I tested the current git version (ea36682cfe4fce5ee65573ba4f5da224abeda487) with a Baumer VLXT-123M.I (10GigE) and it works perfectly.