AravisProject/aravis #230: Not able to create multiple Fake-cameras bound to net.interfaces

niilona, 2019-02-06:
Fake camera can be created only to name "Aravis-GV01".

Also, multiple Fake-cameras failed because bind failed to discovery address

EmmanuelP, 2019-02-07:
Hi Niilona. I'm pasting here your email sent to aravis mailing list. Answer in the next comment.

---

Hi.

Trying to create more than one fake -camera.
Tryin to bind them to physical interfaces like :

- arv-fake-gv-camera-0.6 -i IF1
- arv-fake-gv-camera-0.6 -i IF2

By the 1st one is OK, but others failing cause g_socket_bind returned
"Address already in use"
Am not clear that in case of DGRAM ( = UDP) why this bind is
necessary/possible at all ?
Using latest "git master"
Glib v 2.53.6

# ifconfig IF1
IF1  Link encap:Ethernet  HWaddr 00:1b:21:d4:d2:a0
          inet addr:10.10.3.1  Bcast:10.10.3.255  Mask:255.255.255.0
          inet6 addr: fe80::21b:21ff:fed4:d2a0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18471 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:3833872 (3.8 MB)

# ifconfig IF2
IF2  Link encap:Ethernet  HWaddr 00:1b:21:d4:d2:a0
          inet addr:10.10.4.1  Bcast:10.10.4.255  Mask:255.255.255.0
          inet6 addr: fe80::21b:21ff:fed4:d2a0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:18483 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:3835096 (3.8 MB)

Seems that DGRAM socket bind to address 255.255.255.255 fails in
arvgvfakecamera.c :

```
inet_address = g_inet_address_new_from_string ("255.255.255.255");
discovery_address_string = g_inet_address_to_string (inet_address);
arv_debug_device ("[GvFakeCamera::start] Discovery address = %s",
discovery_address_string);
inet_socket_address = g_inet_socket_address_new (inet_address,
ARV_GVCP_PORT);
if (g_strcmp0 (gvcp_address_string, discovery_address_string) == 0)
    gv_fake_camera->priv->discovery_socket = NULL;
else {
    gv_fake_camera->priv->discovery_socket = g_socket_new
(G_SOCKET_FAMILY_IPV4,
G_SOCKET_TYPE_DATAGRAM,
G_SOCKET_PROTOCOL_UDP, NULL);
if (!g_socket_bind (gv_fake_camera->priv->discovery_socket,
inet_socket_address, FALSE, NULL))
    binding_error = TRUE;
g_socket_set_blocking (gv_fake_camera->priv->discovery_socket, FALSE);
```

Thanks in advance for some hint about

Niilo Minkkinen

EmmanuelP, 2019-02-07:
arv-fake-gv-camera is mostly a debugging tool, and until now I didn't have a use case to more than one instance on a given machine.

I guess you have identified the 2 main issues that prevents more than one instance:

  * Device naming (which should pretty trivial to fix)
  * And interface binding

Interface binding is necessary because device has to listen to the GigEVision control port. Which means the machine should be configured to have multiple IP adresses (Your machine seems correctly configured). The problem in aravis is the use of 255.255.255.255 as the binding address.

I don't plan to work on this issue, but will gladly accept a patch that allows multiple fake-gv-camera instances.

Cheers,

    Emmanuel.

niilona, 2019-02-08:
Thanks Emmanuel

Just working with this and identified mentioned things.
Fixed them and testing, between different machines (working)
Tryin to add own images into (maybe rolling image-buffer).
Will deliver branch soon, have some milestone tested.

Niilo

niilona, 2019-04-11:
Hi Emmanuel.

Just moved that into side as :
- creation of multiple cameras works
- using pre-capured raw's as input works

Now working with DALSA -cameras so busy on in office.

I try to update Fake's as I can into branch I created.

-niilo

EmmanuelP, 2019-08-15:
Hi @niilona 

The current master branch should allow to launch multiple instances of Fake Cameras, with different names.

Cheers.

closed 2019-08-15 by EmmanuelP