AravisProject/aravis #231: properly usage of arv_make_thread_realtime

rickvavilin, 2019-02-07:
How to use arv_make_thread_realtime properly? If I call arv_make_thread_realtime from main thread, only main thread becomes realtime, but I need make realtime internal ArvStream thread.
[image]

EmmanuelP, 2019-02-07:
Hi Rick,

You can change the realtime priority of the stream receiving thread in the stream callback. Please have a look at tests/arvcameratest.c here: https://github.com/AravisProject/aravis/blob/39d58e1295fb3505809eeac1b371f9da946a9d08/tests/arvcameratest.c#L178-L190 and here: https://github.com/AravisProject/aravis/blob/39d58e1295fb3505809eeac1b371f9da946a9d08/tests/arvcameratest.c#L340

Cheers,

    Emmanuel.

closed 2019-02-07 by EmmanuelP

rickvavilin, 2019-02-07:
It's great! Thanks!

commit b41e0c3d (2019-08-02): uvstream: remove parity assertion on alignment

Fixes #231