AravisProject/aravis #233: g_object_unref assertion failed when calling arv_open_device(device_id)

crabsmaps, 2019-03-04:
Problem: I keep seeing following error message when calling arv_open_device(device_id)
`(process:11529): GLib-GObject-CRITICAL **: 10:51:40.389: g_object_unref: assertion 'G_IS_OBJECT (object)' failed`

My setup: I have two basler usb cameras (acA4096-30uc) and I have two applications that should grab particular camera based on "DeviceUserID" field stored inside of the camera. Currently my application grabs every device available until it finds matching "DeviceUserID". I suspect that the error message appears when two applications are trying to attach themselfs simultaneously to the same camera.

Here is a snippet of my code:
```
ArvDevice * getDeviceFromDeviceUserId(std::string desiredDeviceUserID)
{
  ArvDevice * device;
  arv_update_device_list();
  std::string currentDeviceUserID;
  const char*  device_id;
  int nDevices = arv_get_n_devices();
  for (int i = 0; i < nDevices; i++) {
    device_id = arv_get_device_id(i);
    device = arv_open_device(device_id);
    if (!ARV_IS_DEVICE(device)) continue;
    currentDeviceUserID = arv_device_get_string_feature_value(device, "DeviceUserID");
    if (currentDeviceUserID == desiredDeviceUserID) {
      return device;
    }
    else {
      g_object_unref(device);
    }
  }
  return NULL;
}
```

Here is also output with ARV_DEBUG=device:3 (s/n of the cameras have been substituted)
```
[UvDevice::new] Vendor  = Basler
[UvDevice::new] Vendor  = Basler
[UvDevice::new] Product = acA4096-30uc
[UvDevice::new] Product = acA4096-30uc
[UvDevice::new] S/N     = 001
[UvDevice::new] S/N     = 001
[UvDevice::new] Using control endpoint 4, interface 0
[UvDevice::new] Using data endpoint 1, interface 1
Get genicam
[UvDevice::new] Using control endpoint 4, interface 0
[UvDevice::new] Using data endpoint 1, interface 1
[UvDevice::new] Failed to claim USB interface to 'acA4096-30uc - #001'

(process:13925): GLib-GObject-CRITICAL **: 11:44:01.098: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
MANUFACTURER_NAME =        'Basler'
MAX_DEVICE_RESPONSE_TIME = 0x00000032
DEVICE_CAPABILITY        = 0x0000000000000e09
SRBM_ADDRESS =             0x00000000affc0000
MANIFEST_TABLE_ADDRESS =   0x00000000afff0000
[UvDevice::new] Vendor  = Basler
[UvDevice::new] Product = acA4096-30uc
[UvDevice::new] S/N     = 001
U3VCP_CAPABILITY =         0x00000003
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x00000000affd0000
SI_INFO =                  0x02000000
SI_CONTROL =               0x00000000
SI_REQ_PAYLOAD_SIZE =      0x00000000001fa428
SI_REQ_LEADER_SIZE =       0x00000400
SI_REQ_TRAILER_SIZE =      0x00000400
SI_MAX_LEADER_SIZE =       0x00000400
SI_PAYLOAD_SIZE =          0x00100000
SI_PAYLOAD_COUNT =         0x00000001
SI_TRANSFER1_SIZE =        0x000fa428
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000400
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 00 00 00 c0 00 00 00 00  ................
00000010 a0 9e 00 00 00 00 00 00 0e 7d c5 2d c2 c8 64 67  .........}.-..dg
00000020 be 42 dd 24 a2 d9 2f f2 53 a5 11 35 00 00 00 00  .B.$../.S..5....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x00000000c0000000
genicam size    =          0x0000000000009ea0
[UvDevice::new] Using control endpoint 4, interface 0
[UvDevice::new] Using data endpoint 1, interface 1
[UvDevice::new] Failed to claim USB interface to 'acA4096-30uc - #001'

(process:13925): GLib-GObject-CRITICAL **: 11:44:01.104: g_object_unref: assertion 'G_IS_OBJECT (object)' failed
zip file = Basler_Ace_USB_a575e410_Version_1_0.xml
[UvDevice::new] Vendor  = Basler
[UvDevice::new] Product = acA4096-30uc
[UvDevice::new] S/N     = 002
[UvDevice::new] Using control endpoint 4, interface 0
[UvDevice::new] Using data endpoint 1, interface 1
Get genicam
MANUFACTURER_NAME =        'Basler'
MAX_DEVICE_RESPONSE_TIME = 0x00000032
DEVICE_CAPABILITY        = 0x0000000000000e09
SRBM_ADDRESS =             0x00000000affc0000
MANIFEST_TABLE_ADDRESS =   0x00000000afff0000
U3VCP_CAPABILITY =         0x00000003
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x00000000affd0000
SI_INFO =                  0x02000000
SI_CONTROL =               0x00000000
SI_REQ_PAYLOAD_SIZE =      0x00000000004c8028
SI_REQ_LEADER_SIZE =       0x00000400
SI_REQ_TRAILER_SIZE =      0x00000400
SI_MAX_LEADER_SIZE =       0x00000400
SI_PAYLOAD_SIZE =          0x00100000
SI_PAYLOAD_COUNT =         0x00000004
SI_TRANSFER1_SIZE =        0x000c8028
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000400
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 00 00 00 c0 00 00 00 00  ................
00000010 00 ac 00 00 00 00 00 00 9e 9d 39 df 69 a7 f2 15  ..........9.i...
00000020 0a c1 57 5e f0 e9 48 8a e9 c3 31 39 00 00 00 00  ..W^..H...19....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x00000000c0000000
genicam size    =          0x000000000000ac00
zip file = Basler_Ace_USB_9da3a869_Version_1_0.xml
```

While debugging I traced that this error message originates in here https://github.com/AravisProject/aravis/blob/master/src/arvuvdevice.c#L700

crabsmaps, 2019-03-05:
I would imagine that chaging
```	g_clear_object (&uv_device->priv->genicam);```
to
```	if (ARV_IS_GC (uv_device->priv->genicam)) g_clear_object (&uv_device->priv->genicam);```
will fix an issue

crabsmaps, 2019-03-06:
Oh, I was using old version of Aravis
Turns out this PR closes the issue
https://github.com/AravisProject/aravis/pull/215

crabsmaps, 2019-03-06:
Actually since PR was not merged, I will reopen the issue

EmmanuelP, 2019-03-06:
Something similar was pushed to master https://github.com/AravisProject/aravis/blob/57c1d0bc9e3ce92e506cdeab646ae0a3b5a428e3/src/arvuvdevice.c#L729

closed 2019-03-06 by EmmanuelP