AravisProject/aravis #235: It occured a problem about errors in the middle of capturing images

CaffeMrDe, 2019-03-20:
Platform : andriod 4.4
Aravis version : 0.6.1

It occured a problem about errors in the middle of capturing images. The image is wrong, but the collection request is no problem and  the result" (arv_buffer_get_status (buffer) == ARV_BUFFER_STATUS_SUCCESS) " is true. 

Once this error occurs, the camera is wrong.  The result "arv_buffer_get_status (arv_buffer)" is -2.

The following  is :

// pop the buffer from the running stream 
void new_buffer_cb(ArvStream *stream, void* )
{
    boost::mutex::scoped_lock lockGuard(cam->lastBufMut);

    ArvBuffer *arv_buffer= arv_stream_timeout_pop_buffer (stream, 1000);
    // change ArvBuffer to cv::Mat (lastBufMut), including do other something
    // note: lastBufMut is global variable 

    arv_stream_push_buffer (stream, *arv_buffer);
}

ArvBuffer* getPicture()
 {
     {   
        boost::mutex::scoped_lock lockGuard(lastBufMut);
        if (lastBuffer == NULL)
            return ret;
        frame = lastBuffer;
    }
    return ret;
}

Please help me looking for the problem. Thanks.

CaffeMrDe, 2019-03-20:
I set boost::mutex::scoped_lock to ensure atomicity. Thanks

EmmanuelP, 2019-04-06:
Please attach a log of your running application with environment variable ARV_DEBUG set to all:3

EmmanuelP, 2019-09-04:
@CaffeMrDe I'm closing this report for now. Please reopen if you have the requested informations.

Cheers.

closed 2019-09-04 by EmmanuelP