AravisProject/aravis #236: Basler SCA640-120GM cannot exceed 2 FPS

lemmarathon, 2019-03-21:
Basler SCA640-120GM fails to capture faster than 2 fps, when it should be capable of 122 fps at VGA resolution.

Example:
```c
#include <arv.h>
#include <stdio.h>
#include <stdlib.h>
#include <time.h>

int main(int argc, char **argv) {
  ArvCamera *camera = arv_camera_new(NULL);
  if (!camera) return -1;

  arv_camera_set_frame_rate(camera, 30);

  gint payload = arv_camera_get_payload(camera);

  ArvStream *stream = arv_camera_create_stream(camera, NULL, NULL);
  if (!stream) return -1;

  for (int i = 0; i < 50; i++)
    arv_stream_push_buffer(stream, arv_buffer_new(payload, NULL));

  arv_camera_start_acquisition(camera);

  // This loop should take 2 seconds.
  time_t start = time(NULL);
  for (int i = 0; i < 60; i++) {
    ArvBuffer *buf = arv_stream_pop_buffer(stream);
    if (buf) arv_stream_push_buffer(stream, buf);
  }

  // Expected "fps: 30.0", but prints "fps: 2.0".
  time_t elapsed = time(NULL) - start;
  double fps = 60.0 / elapsed;
  printf("fps: %.1f\n", fps);

  arv_camera_stop_acquisition(camera);
  g_object_unref(stream);
  g_object_unref(camera);
}
```

Running the above snippet results in:
```
Got fps: 2.0
```

Expected:
```
Got fps: 30.0
```

EmmanuelP, 2019-03-22:
Hi,

Did you check exposure time is small enough ( < 33 ms ) ?

Could you attach the output of `tests/arv-camera-test -f 30 - e 30000 -d all`, please ?

lemmarathon, 2019-03-23:
Exposure time was set as low as 10 ms with same error.

Output of `tests/arv-camera-test -f 30 - e 30000 -d all`:
[arv-camera-test.log](https://github.com/AravisProject/aravis/files/2998815/arv-camera-test.log)

EmmanuelP, 2019-03-23:
Looking at the log file, packet transmission from the camera to the machine works fine.
It is probably an issue when aravis tries to set up the acquisition frame rate. What you could try now is:
- If it is possible, use pylon to configure the acquisition frame rate, exit Pylon and launch arv-camera-test without any parameter. It should work with the same frame rate as with Pylon.
- Look at the camera documentation and check if the instructions for frame rate setting corresponds to what is done in aravis: https://github.com/AravisProject/aravis/blob/57c1d0bc9e3ce92e506cdeab646ae0a3b5a428e3/src/arvcamera.c#L872

lemmarathon, 2019-03-30:
We ended up switching to a different camera, but feel free to leave this issue open in case anyone decides to troubleshoot this model in the future.