AravisProject/aravis #237: Receiving incorrect value of float feature

crabsmaps, 2019-03-25:
My camera is Basler acA4096-30uc.
Executing command `arv-tool-0.6 control DeviceTemperature` returns `DeviceTemperature = 0 C`

If I'm querying pValue directly `arv-tool-0.6 control N906` I get correct reading `N906 = 57`

Relevant part of genicam.xml
```
  <Float Name="DeviceTemperature" NameSpace="Standard">
    <ToolTip>Temperature of the selected location within the device (in degrees centigrade).</ToolTip>
    <Description>Temperature of the selected location within the device (in degrees centigrade). The temperature is measured at the location set by DeviceTemperatureSelector.</Description>
    <DisplayName>Device Temperature</DisplayName>
    <Visibility>Expert</Visibility>
    <pValue>N904</pValue>
    <Unit>C</Unit>
    <Representation>PureNumber</Representation>
    <DisplayNotation>Fixed</DisplayNotation>
    <DisplayPrecision>1</DisplayPrecision>
  </Float>
  <Integer Name="N904" NameSpace="Custom">
    <pIsImplemented>N903</pIsImplemented>
    <pIsAvailable>N905</pIsAvailable>
    <pValue>N906</pValue>
  </Integer>
  <MaskedIntReg Name="N905" NameSpace="Custom">
    <Address>0x11000</Address>
    <pAddress>N907</pAddress>
    <Length>4</Length>
    <AccessMode>RO</AccessMode>
    <pPort>Device</pPort>
    <Bit>30</Bit>
    <Sign>Unsigned</Sign>
    <Endianess>LittleEndian</Endianess>
  </MaskedIntReg>
  <IntReg Name="N906" NameSpace="Custom">
    <Address>0x11008</Address>
    <pAddress>N907</pAddress>
    <Length>4</Length>
    <AccessMode>RO</AccessMode>
    <pPort>Device</pPort>
    <Cachable>NoCache</Cachable>
    <PollingTime>30000</PollingTime>
    <Sign>Signed</Sign>
    <Endianess>LittleEndian</Endianess>
  </IntReg>
```

This is really strange since DeviceTemperature feature is declared to be float, but underneath is just a Integer.
I suppose relevant piece of logic is in here https://github.com/AravisProject/aravis/blob/master/src/arvgcpropertynode.c#L403
Since underlying property node is of Integer type this function returns 0.0 directly.

EmmanuelP, 2019-03-25:
What is the output of `arv-tool-0.6 control DeviceTemperature -d device:3,genicam:3` ?

crabsmaps, 2019-03-26:
output is quite huge, I'll paste relevant parts, but let me know if you would like to see full output
```
[UvDevice::new] Vendor  = Basler
[UvDevice::new] Product = acA4096-30uc
[UvDevice::new] S/N     = xxxxxxxx
[UvDevice::new] Using control endpoint 4, interface 0
[UvDevice::new] Using data endpoint 1, interface 1
Get genicam
MANUFACTURER_NAME =        'Basler'
MAX_DEVICE_RESPONSE_TIME = 0x00000032
DEVICE_CAPABILITY        = 0x0000000000000e09
SRBM_ADDRESS =             0x00000000affc0000
MANIFEST_TABLE_ADDRESS =   0x00000000afff0000
U3VCP_CAPABILITY =         0x00000003
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x00000000affd0000
SI_INFO =                  0x02000000
SI_CONTROL =               0x00000000
SI_REQ_PAYLOAD_SIZE =      0x0000000000878000
SI_REQ_LEADER_SIZE =       0x00000400
SI_REQ_TRAILER_SIZE =      0x00000400
SI_MAX_LEADER_SIZE =       0x00000000
SI_PAYLOAD_SIZE =          0x00000000
SI_PAYLOAD_COUNT =         0x00000000
SI_TRANSFER1_SIZE =        0x00000000
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000000
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 00 00 00 c0 00 00 00 00  ................
00000010 00 ac 00 00 00 00 00 00 9e 9d 39 df 69 a7 f2 15  ..........9.i...
00000020 0a c1 57 5e f0 e9 48 8a e9 c3 31 39 00 00 00 00  ..W^..H...19....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x00000000c0000000
genicam size    =          0x000000000000ac00
zip file = Basler_Ace_USB_9da3a869_Version_1_0.xml
[Gc::register_feature_node] Register node 'Root' [Category]

[...]

[Gc::register_feature_node] Register node 'DeviceLinkCurrentThroughput' [IntReg]
[Gc::register_feature_node] Register node 'DeviceTemperatureSelector' [Enumeration]
[Gc::register_feature_node] Register node 'DeviceTemperature' [Float]
[Gc::register_feature_node] Register node 'N904' [Integer]
[Gc::register_feature_node] Register node 'N905' [MaskedIntReg]
[Gc::register_feature_node] Register node 'N906' [IntReg]
[Gc::register_feature_node] Register node 'N907' [IntSwissKnife]
[Gc::register_feature_node] Register node 'N908' [MaskedIntReg]
[Gc::register_feature_node] Register node 'N903' [IntSwissKnife]
[Gc::register_feature_node] Register node 'TemperatureState' [Enumeration]
[Gc::register_feature_node] Register node 'N911' [MaskedIntReg]
[Gc::register_feature_node] Register node 'N912' [IntReg]
[Gc::register_feature_node] Register node 'N913' [MaskedIntReg]
[Gc::register_feature_node] Register node 'N914' [IntReg]

[...]

Basler-xxxxxxxx (USB3)
[GcPropertyNode::get_double] Invalid node 'N904'
DeviceTemperature = 0 C
```

crabsmaps, 2019-03-26:
Here is a full output of `arv-tool-0.6 control DeviceTemperature -d device:3,genicam:3`
[basler-debug.txt](https://github.com/AravisProject/aravis/files/3008835/basler-debug.txt)

EmmanuelP, 2019-04-01:
Please test https://github.com/AravisProject/aravis/pull/240

closed 2019-04-02 by EmmanuelP