AravisProject/aravis #238: Unable to change exposure/gain camera settings without power cycling camera

deroliver, 2019-03-27:
I am using a Basler ACA4600-7GC to get acquire single images via the _arv_camera_acquisition_ function.

For my current use case, I need to:

1. connect to my camera
2. set the exposure/gain
3. acquire an image
4. change the exposure/gain
5. acquire a second image with the new exposure/gain settings

For step 4, I am using the functions _arv_camera_set_exposure_time_ and _arv_camera_set_gain_, with the ArvCamera object and the value as arguments.

When I acquire a second image during step 5, the camera still used the original exposure/gain values.

I am unable to change the exposure and gain, and only when I power cycle the camera does the camera use the updated exposure/gain settings from step 4.

Is there an additional step to updating these values on the camera that I am missing, or a function call I need to make?
I have also tried:

1. updating the exposure/gain
2. closing the camera connection
3. reconnecting to the camera 
4.acquiring an image

The camera still will not use the updated exposure/gain settings.

Thanks for the help!

EmmanuelP, 2019-04-06:
Please attach a log of your running application with environment variable `ARV_DEBUG` set to `all:3`

deroliver, 2019-04-08:
Thank you for the reply! Here is the attached log file.
[cam.log](https://github.com/AravisProject/aravis/files/3054796/cam.log)

EmmanuelP, 2019-04-11:
I did not find anything interresting in the log file.
Did you succeed to use arv-viewer ? Is the gain updated when you move the slide in the setting drop down menu ?

EdCha, 2019-07-14:
I am running a Basler as well, Aviator 2300, with Qarv.

The following may or may not be related, but under Qarv master Branch the gain and exposure sliders are locked up.

Jure Valrec (Qarv developer) has created a branch called "edouard". I quote Jure here 

"Back then, it was Basler's fault, the camera only defined ExposureRaw but not ExposureAbs (or 
something like that). Maybe that is the case with your current camera as well?"

So Jure modified the Edouard branch to catch up with Qt5 and opencv 4.0
However I am still facing a segfault with

[ 8790.213388] QArv Cooker[26472]: segfault at 7efd9056c084 ip 00007efd880dd450 sp 00007efd64859380 error 4 in libopencv_imgproc.so.4.0.0[7efd87f0a000+1449000]

Not sure what it is or if it is relevant or not to the gain / exposure being locked up
Aravis is the latest stable release with opencv 4.0

Ed

EmmanuelP, 2019-07-15:
Hi Ed,

It is better to open a new issue. And attach there a log file obtained with `ARV_DEBUG` set to `all:3`

Cheers,

    Emmanuel.

EdCha, 2019-07-16:
Bonjour :-)
Sorry for the "noise" on this thread, I realize my comment did not really apply to this particular problem. I will remove it. I am also not too sure about the libopencv_imgproc error ? ist Qarv ? Is it Aravis ? is it OpenCV ?
Thanks
Edouard

EmmanuelP, 2019-09-04:
@deroliver I'm closing this issue. Don't hesitate to reopen if you have the requested informations.

Cheers.

closed 2019-09-04 by EmmanuelP