AravisProject/aravis #242: Two cameras using local-link

hinxx, 2019-04-16:
I have two identical GigE cameras. The Linux host is packing an Intel two port NIC that has jumbo packet support. I configured the both Ethernet interfaces to be link-local, so are the cameras.
I can only see one camera using arv-tool. Is this expected in this configuration?

NIC p1p1 and p1p2 are used dfor the cameras.

```
iocuser@npm-ipc-02: ~$ ifconfig 

p1p1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 8220
        inet 169.254.58.10  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::279d:a666:cece:f188  prefixlen 64  scopeid 0x20<link>
        ether b4:96:91:10:af:f0  txqueuelen 1000  (Ethernet)
        RX packets 8072  bytes 4455360 (4.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 21967  bytes 1659053 (1.5 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0xdf100000-df1fffff  

p1p2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 8220
        inet 169.254.54.133  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::bc00:fabc:bbf5:cb95  prefixlen 64  scopeid 0x20<link>
        ether b4:96:91:10:af:f1  txqueuelen 1000  (Ethernet)
        RX packets 502  bytes 275896 (269.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 764  bytes 103825 (101.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0xdf000000-df0fffff  

p6p1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.30.150.42  netmask 255.255.255.0  broadcast 172.30.150.255
        inet6 fe80::772e:8728:894d:d6ba  prefixlen 64  scopeid 0x20<link>
        ether 64:00:6a:ff:ef:1f  txqueuelen 1000  (Ethernet)
        RX packets 1231752  bytes 438443681 (418.1 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 29018503  bytes 1867028535 (1.7 GiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device memory 0xdf300000-df37ffff  

```

```
iocuser@npm-ipc-02: ~$ /run/media/iocuser/ext500/BDE/out/R3.15.5/artifacts/aravis-master/bin/arv-tool-0.6 --debug all:2
Found 0 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.58.10
[GvDiscoverSocket::new] Add interface 169.254.54.133
[GvDiscoverSocket::new] Add interface 172.30.150.42
[GvDiscoverSocket::new] Add interface 192.168.122.1
[GvInterface::discovery] Device 'Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-08409' found (interface 169.254.58.10) user_id '' - MAC '00:0f:31:02:c2:77'
Found 0 USB3Vision device (among 7 USB devices)
[GvDevice::new] Interface address = 169.254.58.10
[GvDevice::new] Device address = 169.254.20.123
[GvDevice::load_genicam] xml url = 'Local:prosilica_022622A_1.54.xml;100000;3BF1C' at 0x200
[GvDevice::load_genicam] Xml address = 0x100000 - size = 0x3bf1c - prosilica_022622A_1.54.xml
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[... 333 lines cut ...]
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[ArvDevice::set_string_feature_value] Node 'GevGVSPExtendedIDMode' is not a string
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::new] Register workaround = yes
Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-08409 (169.254.20.123)
```

EmmanuelP, 2019-04-16:
No, this is not expected.

In your log, only the camera connected to p1p1 has answered to the discovery request.
Did you check the camera one by one, connected to p1p1 ?

hinxx, 2019-04-16:
>  Did you check the cameras one by one, connected to p1p1 ?
I did try connecting one then the other camera to one of the ports earlier (not sure which one) and it worked both times - I got the camera to answer.

I can disable p1p1 interface and get the other one to answer:

```
iocuser@npm-ipc-02: ~$ /run/media/iocuser/ext500/BDE/out/R3.15.5/artifacts/aravis-master/bin/arv-tool-0.6 --debug all:2
Found 0 USB3Vision device (among 7 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.54.133
[GvDiscoverSocket::new] Add interface 172.30.150.42
[GvDiscoverSocket::new] Add interface 192.168.122.1
[GvInterface::discovery] Device 'Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-10012' found (interface 169.254.54.133) user_id '' - MAC '00:0f:31:03:27:14'
Found 0 USB3Vision device (among 7 USB devices)
[GvDevice::new] Interface address = 169.254.54.133
[GvDevice::new] Device address = 169.254.43.40
[GvDevice::load_genicam] xml url = 'Local:prosilica_022622A_1.54.xml;100000;3BF1C' at 0x200
[GvDevice::load_genicam] Xml address = 0x100000 - size = 0x3bf1c - prosilica_022622A_1.54.xml
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (pSelected)
[Genicam::create_element] Unknow tag (pSelected)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Streamable)
[Genicam::create_element] Unknow tag (Streamable)
[... 297 lines cut ...]
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[ArvDevice::set_string_feature_value] Node 'GevGVSPExtendedIDMode' is not a string
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::new] Register workaround = yes
Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-10012 (169.254.43.40)
```

hinxx, 2019-04-17:
Anything I can help with debugging thos one?

EmmanuelP, 2019-04-18:
Well, aravis is supposed to send a discovery packet on each interface listed here:

```
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.58.10
[GvDiscoverSocket::new] Add interface 169.254.54.133
[GvDiscoverSocket::new] Add interface 172.30.150.42
[GvDiscoverSocket::new] Add interface 192.168.122.1
```
And wait during 1 second for incoming discovery acknowledges.

So, you could watch the ethernet interfaces using wireshark and look at the packets. There is a GigeVision dissector that helps to identify the interesting ones.

Also, you may have a non standard network setup. How are connected the cameras ?

niilona, 2019-04-18:
Hi.

Looked your network settings and two comments on :

1.) Suggest you to use fixed IP's or at least tighten netmask to
255.255.255.0
2.) I have had difficulties on cases when cameras "see" others traffic and
therefore isolated them using VLAN's (maybe not possible in your case)

Have experienced with Aravis - and other manufacturer specific libs with
multi-camera systems.

Cheers

-niilo

hinxx, 2019-04-23:
> 1.) Suggest you to use fixed IP's or at least tighten netmask to 255.255.255.0

@niilona, I want to avoid using static IPs, in the first place. The commodity of link-local looks attractive and feasible.

I looked for the place where netmask setting for link-local would be present, on Centos Linux, but could not find anything. Searching the web didn't help either. If I just set the netmask to 255.255.255.0 after the link-local IP has been negotiated between the PC and camera, I lose the communication to that camera.

@EmmanuelP, my setup is single Linux PC, one NIC with two ports, two identical cameras hooked up.

I could look into VLANs, but I fear (by not knowing how to deal with VLANs) that this might be a even more complex than just start using static IPs (which I would really like to avoid).

I guess having a local DHCP server would cut it, but again the configuration and maintanence of the DHCP server is another item that might be more complex than using the static IPs ..

Link-local looks really attractive and works like a charm with one camera.

Will fire up the wireshark and look at traffic.

hinxx, 2019-04-23:
Using avahi-autoipd (instead of using NetworkManager UI; both p1p1 and p1p2 are ignored by the NM), I got these IPs on Linux box:

```
p1p1:avahi: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 8220
        inet 169.254.10.85  netmask 255.255.0.0  broadcast 169.254.255.255
        ether b4:96:91:10:af:f0  txqueuelen 1000  (Ethernet)
        device memory 0xdf100000-df1fffff  

p1p2:avahi: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 8220
        inet 169.254.10.90  netmask 255.255.0.0  broadcast 169.254.255.255
        ether b4:96:91:10:af:f1  txqueuelen 1000  (Ethernet)
        device memory 0xdf000000-df0fffff  

```

Wireshark traffic capture for of the single arv-tool-0.6 invocation results in this output (only one camera so far) `Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-08409 (169.254.20.123) `:

[image]

You can see both pings and pongs, and looking at the data in the responses I can tell that both cameras answered:

[image]

[image]

But only one was asked for its MAC using ARP. There is no second ARP request/response in the rest of the wireshark capture (filtered by ARP type):

[image]

I hope this sheds some light onto the issue.

hinxx, 2019-04-23:
Using this command:
```
iocuser@npm-ipc-02: ~$ /tmp/bin/arv-tool-0.6 -a 169.254.20.123
169.254.20.123
```
I get a response from one camera, but not the other:
```
iocuser@npm-ipc-02: ~$ /tmp/bin/arv-tool-0.6 -a 169.254.43.40
Device '169.254.43.40' not found
```

In the meantime wireshark recorded this:

[image]

The ARP packets were issued this time for the second camera, but no-one responded. After no ARP responses, the ping/pong for both cameras appear.

hinxx, 2019-04-24:
I had a chance today to try out the static IPs, with tighter netmask, as suggested (I used 255.255.255.0). I still haven't figured out how to set netmask for link-local.

Linux:
```
p1p1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 8228
        inet 192.5.96.1  netmask 255.255.255.0  broadcast 192.5.96.255
        ether b4:96:91:10:af:f0  txqueuelen 1000  (Ethernet)

p1p2: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 8228
        inet 192.5.95.1  netmask 255.255.255.0  broadcast 192.5.95.255
        ether b4:96:91:10:af:f1  txqueuelen 1000  (Ethernet)
```

Cameras:
```
iocuser@npm-ipc-02: x64$ /tmp/bin/arv-tool-0.6 
Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-08409 (192.5.96.96)
Allied Vision Technologies-GT3300 (02-2622A)-02-2622A-10012 (192.5.95.95)
```

This works.
If I go back and use the 255.255.0.0 netmask, even with the static IPs, I only see a single camera.

hinxx, 2019-04-25:
Forgot to mention that I experience the same behavior when using Prosilica PvAPI CLI tool to locate the cameras; IOW, the issue does not seem directly related to aravis only.

I would still like to understand if the only reason for link-local not working lies in the 255.255.0.0 netmask, or there exists a more subtle reason for this.

EmmanuelP, 2019-07-24:
So the whole issue seems to be a network configuration issue, not something that has to be fixed in aravis.

I'm cosing this issue. Don't hesitate to reopen if you think something in aravis prevents you to make your setup work.

Cheers,

  Emmanuel.

closed 2019-07-24 by EmmanuelP