AravisProject/aravis #243: Gstreamer Plugin, some formats not available.

sg-brianvg, 2019-04-18:
I have connected a Basler acA1920-40gm/gc camera on an embedded ARM platform. When I use the Aravissrc Gstreamer1.0 plugin, I can only seem to negotiate either Bayer or Grayscale resolutions. Whenever I try to use YUV 4:2:2 Packed or YUV 4:2:2 (YUYV) Packed, both of which are, according to the Basler documentation, supported pixel formats, I get a "caps not negotiated" error from Gstreamer. I can get color video from the camera using a Bayer format output, but translating this into a pixel format for compression is computationally a bit expensive. It would be great if I could get the YUV modes working.

Please let me know what details I can send you to clarify the issue, and thanks for your great work.

sg-brianvg, 2019-04-18:
BTW, the Basler documentation is here:

https://graftek.biz/system/files/3086/original/Basler_Ace_GigE_Manual.pdf?1479132086

EmmanuelP, 2019-04-18:
I have just tested the viewer, and both YUV 4:2:2 Packed and YUV 4:2:2 (YUYV) Packed are correctly displayed:

[image]

May be it is a broken gstreamer installation ? What you could try would be to create a gstreamer pipeline with  aravissrc replaced by a videotestsrc element, and make videotestsrc emit a YUV stream.

sg-brianvg, 2019-04-23:
Hi Emmanual,

That is very perceptive. Indeed, when I try videotestsrc based pipelines, I also can't seem to get YU(Y)V to work, with gstreamer claiming that videotestsrc can't handle the caps string. Now I guess I need to find out what is wrong with the Xilinx Petalinux Gstreamer-1.0 plugin installation recipes.

Thanks for your help!

Brian

sg-brianvg, 2019-04-23:
JFYI, the complete pipeline looks like this:
`gst-launch-1.0 aravissrc camera-name=Basler-acA1920-50gc-22372944 \
                  ! 'video/x-raw,width=640,height=480,framerate=15/1,format=UYVY' \
                  ! queue \
                  ! omxh264enc num-slices=8 gop-length=60 periodicity-idr=60 \
                    control-rate=low-latency target-bitrate=5000 cpb-size=3000 prefetch-buffer=true \
                  ! video/x-h264, profile=high ! queue ! rtph264pay ! udpsink host=192.168.21.86 port=50000 max-lateness=-1 qos-dscp=60 async=false buffer-size=60000000 max-bitrate=120000000
`

EmmanuelP, 2019-07-24:
So it looks like it is not something aravis is responsible for, more a gstreamer pipeline configuration.

I'm closing the issue. Please reopen if you think somtehing in aravis is preventing you to use the aravis video stream in a gstreamer pipeline.

Cheers,

    Emmanuel.

closed 2019-07-24 by EmmanuelP