AravisProject/aravis #245: Cross compilation for ARM using GigE camera and Aravis

payala07, 2019-05-07:
Hello,
My issue is the following, I'm using a camera Basler acA1300-30gm alongside with Aravis. It works well on the local system and also while calling from a different network using uniscast. On the next step of my experiment I require to cross compile Aravis in order to use on an ARM system (modified armv7l).
Using the configure file :
[image]
I didn't have any problem creating the executables:
[image]
The problem though is that, to be able to make it run on the ARM system I required it to be ELF 32-bits instead of ELF-64 bits. The following is the error given:
[image]
[image]
[image]
If I try cross-compiling exclusively the arv-camera-test after the ./configure worked, I get the following error:
[image]
arm-buildroot-linux-uclibcgnueabi-gcc: /root/root/aravis/lib/libaravis-0.6.a: No such file or directory
In file included from /root/aravis/include/aravis-0.6/arvtypes.h:30,
                 from /root/aravis/include/aravis-0.6/arv.h:28,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvenums.h:30:25: error: glib-object.h: No such file or directory
In file included from /root/aravis/include/aravis-0.6/arvtypes.h:30,
                 from /root/aravis/include/aravis-0.6/arv.h:28,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvenums.h:34: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'typedef'
/root/aravis/include/aravis-0.6/arvenums.h:88: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'ArvPixelFormat'
In file included from /root/aravis/include/aravis-0.6/arv.h:28,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvtypes.h:33: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
In file included from /root/aravis/include/aravis-0.6/arv.h:31,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvversion.h:32: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
In file included from /root/aravis/include/aravis-0.6/arv.h:32,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvdebug.h:30:18: error: glib.h: No such file or directory
In file included from /root/aravis/include/aravis-0.6/arv.h:32,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvdebug.h:40: warning: data definition has no type or storage class
/root/aravis/include/aravis-0.6/arvdebug.h:44: error: expected specifier-qualifier-list before 'ArvDebugLevel'
/root/aravis/include/aravis-0.6/arvdebug.h:112: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_debug_check'
In file included from /root/aravis/include/aravis-0.6/arv.h:33,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvmisc.h:32: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
/root/aravis/include/aravis-0.6/arvmisc.h:36: error: expected '=', ',', ';', 'asm' or '__attribute__' before '*' token
/root/aravis/include/aravis-0.6/arvmisc.h:37: error: expected ')' before '*' token
/root/aravis/include/aravis-0.6/arvmisc.h:38: error: expected ')' before '*' token
/root/aravis/include/aravis-0.6/arvmisc.h:39: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_statistic_fill'
/root/aravis/include/aravis-0.6/arvmisc.h:41: error: expected ')' before '*' token
/root/aravis/include/aravis-0.6/arvmisc.h:43: error: expected ';', ',' or ')' before '*' token
/root/aravis/include/aravis-0.6/arvmisc.h:47: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_value_get_type'
/root/aravis/include/aravis-0.6/arvmisc.h:51: error: expected specifier-qualifier-list before 'GType'
/root/aravis/include/aravis-0.6/arvmisc.h:62: error: expected declaration specifiers or '...' before 'gint64'
/root/aravis/include/aravis-0.6/arvmisc.h:64: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_value_get_int64'
/root/aravis/include/aravis-0.6/arvmisc.h:66: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_value_holds_int64'
/root/aravis/include/aravis-0.6/arvmisc.h:69: error: expected declaration specifiers or '...' before 'size_t'
/root/aravis/include/aravis-0.6/arvmisc.h:69: error: expected declaration specifiers or '...' before 'guint'
/root/aravis/include/aravis-0.6/arvmisc.h:70: error: expected declaration specifiers or '...' before 'size_t'
/root/aravis/include/aravis-0.6/arvmisc.h:70: error: expected declaration specifiers or '...' before 'guint'
/root/aravis/include/aravis-0.6/arvmisc.h:72: error: expected declaration specifiers or '...' before 'size_t'
/root/aravis/include/aravis-0.6/arvmisc.h:72: error: expected declaration specifiers or '...' before 'size_t'
/root/aravis/include/aravis-0.6/arvmisc.h:74: error: expected ')' before 'pixel_format'
/root/aravis/include/aravis-0.6/arvmisc.h:75: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_pixel_format_from_gst_caps'
/root/aravis/include/aravis-0.6/arvmisc.h:76: error: expected ')' before 'pixel_format'
/root/aravis/include/aravis-0.6/arvmisc.h:77: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_pixel_format_from_gst_0_10_caps'
In file included from /root/aravis/include/aravis-0.6/arv.h:34,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvstr.h:33: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
/root/aravis/include/aravis-0.6/arvstr.h:37: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_str_is_uri'
/root/aravis/include/aravis-0.6/arvstr.h:40: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_str_parse_double'
/root/aravis/include/aravis-0.6/arvstr.h:78: error: expected ')' before '*' token
In file included from /root/aravis/include/aravis-0.6/arv.h:35,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvzip.h:32: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
/root/aravis/include/aravis-0.6/arvzip.h:36: error: expected declaration specifiers or '...' before 'size_t'
[... 591 lines cut ...]
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvgvfakecamera.h:32: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
/root/aravis/include/aravis-0.6/arvgvfakecamera.h:45: error: expected specifier-qualifier-list before 'GObject'
/root/aravis/include/aravis-0.6/arvgvfakecamera.h:51: error: expected specifier-qualifier-list before 'GObjectClass'
/root/aravis/include/aravis-0.6/arvgvfakecamera.h:54: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_gv_fake_camera_get_type'
/root/aravis/include/aravis-0.6/arvgvfakecamera.h:57: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_gv_fake_camera_start'
In file included from /root/aravis/include/aravis-0.6/arv.h:88,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvrealtime.h:10: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
/root/aravis/include/aravis-0.6/arvrealtime.h:13: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_make_thread_high_priority'
In file included from /root/aravis/include/aravis-0.6/arv.h:89,
                 from arvacquisitiontest.c:1:
/root/aravis/include/aravis-0.6/arvxmlschema.h:34: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'G_BEGIN_DECLS'
/root/aravis/include/aravis-0.6/arvxmlschema.h:38: warning: data definition has no type or storage class
/root/aravis/include/aravis-0.6/arvxmlschema.h:52: error: expected specifier-qualifier-list before 'GObject'
/root/aravis/include/aravis-0.6/arvxmlschema.h:59: error: expected specifier-qualifier-list before 'GObjectClass'
/root/aravis/include/aravis-0.6/arvxmlschema.h:62: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_xml_schema_get_type'
/root/aravis/include/aravis-0.6/arvxmlschema.h:64: error: expected ')' before '*' token
/root/aravis/include/aravis-0.6/arvxmlschema.h:66: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'arv_xml_schema_validate'
arvacquisitiontest.c:2:20: error: stdlib.h: No such file or directory
arvacquisitiontest.c:3:19: error: stdio.h: No such file or directory
arvacquisitiontest.c:5: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'int'

Is there any version of Aravis that allows me to create the executables under that format? or is a problem of changing the cross compiling path giving all the shared libraries?
I thanks you in advance for your help.

EmmanuelP, 2019-05-13:
Hi,

I have almost no experience in cross compilation. All I know is the cross compilation setup can be tricky to get it working. If possible, I would try to compile natively on the target platform.

payala07, 2019-05-13:
Hello, 
Thanks for your answer Emmanuel, yes, it is indeed quite tricky. So far I was able to fix most of the compilation errors, but it is no perfect yet, so far the library that is stopping it, is related to the time stamp. I will leave here what I did, in case anyone is in a similar spot.
[image]
[image]
Some of the compilation errors:
[image]
[image]

Cheers,
Paola.

payala07, 2019-05-14:
Hello, 
Correcting the line for the cross-compilation:
[image]
That corrects the compilation errors leaving only this one:
[image]
Cheers,
Paola.

payala07, 2019-05-27:
Hello, I made some progress although i'm still stuck. The compiler path for the cross-compilation is the following:
[image]
However, I still got problems with the linker:
[image]
If anyone got an idea of how to solve it, I will be more than thankful.
Cheers,
Paola.

payala07, 2019-05-28:
Hello, some news in the topic, using a Makefile I was able to go around some of the configuration errors, but still have a problem with the linking.
[image]
If someone have some insight, I will be really thankful.
Cheers,
Paola.
P.d.: Here is the Makefile.
[Makefile.txt](https://github.com/AravisProject/aravis/files/3227491/Makefile.txt)

payala07, 2019-05-29:
Hello there, I got some updates:
changing the Makefile for the cross-compiler a little bit:
[Makefile.txt](https://github.com/AravisProject/aravis/files/3231749/Makefile.txt)
the files compiles and links:
[image]
[image]
However it stills generates two problems:
- /root/armel/usr/lib/arm-linux-gnueabi/libc.a(dl-tsd.o)(.text+0x18): R_ARM_TLS_LE32 relocation not permitted in shared object
-:tests/test_paola/arv_acquisition_test # ./arv-acquisition-test
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
Segmentation fault                                                                                       
I know that the first error is due to the contrast in between -static and -shared, however I do not know from where comes the second error. If someone have any idea how to solved it, I will be really grateful.
Cheers,
Paola

payala07, 2019-06-14:
Hello,
I verified it with some colleagues, under the current system this is as far that the cross-compilation goes. It is able to compile, link and generate executable but due to the custom system, the previously mentioned errors appears due to incompatibility.
I will clean up the Makefile for it nevertheless (after cleaning it), it might be of some use for those who are trying to cross-compile Aravis elsewhere.
Cheers,
Paola

closed 2019-06-14 by payala07