AravisProject/aravis #247: Can't create fake camera with Blackfly genicam data

lpea, 2019-05-13:
Hi!

I'm trying to create a fake camera using the genicam data from a FLIR Blackfly ethernet camera. Exact model is BFLY-PGE-31S4M. Aravis version is 0.6.2 taken from ftp.gnome.org.

The genicam data was retrieved from the camera with the following command:
`./arv-tool-0.6 -a <IP> genicam > blackfly.xml`

Then a fake camera is started like so:
`./arv-fake-gv-camera-0.6 -g blackfly.xml`

So far so good, the problem arises when I want to visualize the stream from the fake camera:
`./arv-viewer`

And as soon as I select the fake camera from the list, arv-driver freezes and arv-fake-gv-camera-0.6 crashes with the following error:

> ...
[GvFakeCamera::handle_control_packet] New controller
[GvFakeCamera::thread] Control packet received
[GvFakeCamera::handle_control_packet] Read register command 2356 -> 0
[GvFakeCamera::thread] Control packet received
[GvFakeCamera::handle_control_packet] Read memory command 72 (32)
[GvFakeCamera::thread] Control packet received
[GvFakeCamera::handle_control_packet] Read memory command 104 (32)
[GvFakeCamera::thread] Control packet received
>
>(arv-fake-gv-camera-0.6:29768): Aravis-CRITICAL **: 10:25:02.832: arv_fake_camera_write_memory: assertion 'address + size < ARV_FAKE_CAMERA_MEMORY_SIZE + camera->priv->genicam_xml_size' failed
[GvFakeCamera::handle_control_packet] Write register command -252685312 -> 0
[GvFakeCamera::thread] Control packet received
[1]    29768 segmentation fault (core dumped)  ./arv-fake-gv-camera-0.6 -g blackfly.xml -d all

Is the workflow OK or do I miss something? I tried to dive a bit in the code to understand what exactly caused the error but couldn't find anything so far.

As a side node, I have no problem visualizing the stream from the camera using the viewer.

TIA,
Guillaume

EmmanuelP, 2019-05-13:
Hi Guillaume,

What you are trying to do is not possible.

A genicam xml file describes the registers of a device, it is not possible to use a random file for the fake device, like it is not possible to use the Genicam file from a Basler camera for a PointGrey camera.

The -g genicam option of arv-fake-gv-camera-0.6 is there for testing purpose.

Cheers,

    Emmanuel.

closed 2019-05-13 by EmmanuelP

lpea, 2019-05-13:
Hi Emmanuel,

Thanks for the prompt answer!

My mistake then. I thought the -g option could be used to simulate the behavior of any camera, based on the genicam data, so an actual camera driver would be able to communicate with the simulated one without knowing it's a fake. Some kind of "reverse driver".

Do you think that would be feasible? I wonder if this could be done using Aravis somehow or if it would require to develop something from scratch.

Cheers,
Guillaume

EmmanuelP, 2019-05-13:
It is feasible, but depending on the compatibility you want to achieve, it can be a lot of work.

Nothing is in place in aravis to really emulate a device, but it is doable in aravis, as long as you want to write C code. ArvFakeCamera should be converted to an interface, what is in arvfakecamera.c should be used to write an implementation of this interface for the current fake device, and a new implementation that emulates a real camera should be written.

lpea, 2019-05-13:
I see, many thanks for the pointers.