AravisProject/aravis #252: ARV_BUFFER_STATUS_SUCCESS

jpvlerbe, 2019-06-07:
Hi

I am using the Python bindings for Aravis and when I check a popped buffer a lot of the frames have status: ARV_BUFFER_STATUS_TIMEOUT. I was wondering what can be causing this timeout. The mtu is at 8192, the interface is gigabit and directly connected. 
I have used stream.set_property("packet-timeout",...) to double the default value which did not help.
Increasing the number of buffers did not work.

However when I removed the saving of the image in the buffer to disk, the timeouts were gone. But clearly, that is no solution. Is it caused by other processes swapping in? I have increased the niceness of the python process to -20 which also did not help.

How can i further debug these timeouts?

EmmanuelP, 2019-06-07:
Your program probably blocks the receiving thread. Could you post a small test that shows this issue ?

jpvlerbe, 2019-06-12:
Hi

I have made a MWE that is attached here.
[mwe.py.txt](https://github.com/AravisProject/aravis/files/3281534/mwe.py.txt)

I am running this on an Odroid XU4, which is of course limited in power. With the framerate set to 5 I have no timeouts. With framerate at 10, I have 21 timeouts and acquisition of 100 images takes 17 seconds (cpu load only at 0.4).

EmmanuelP, 2019-07-24:
You code doesn't block the stream receiving thread, but if the processing of frame takes too long, you may run out of buffers in the thread input queue.

If you set the number of buffers to 100, does it fixes the timeout ?

A CPU load of 0.4 (I guess 40%) does not necessarely indicates you have enough ressources, data saving can take time without using CPU.

EmmanuelP, 2019-09-04:
I'm closing this report, as it is not a really specific issue.

Performance can be improved in several ways.

If it is a GigEVision camera:

  - Try to use packet socket support
  - Try to increase ethernet packet size (and also change interface MTU)

In your application, try to use all of your cores.

Make sure the bandwidth required for the video stream fits in your interface bandwidth.

If you have other performance releated questions, please open a topic on the Aravis discourse instance: https://aravis-project.discourse.group/c/using-aravis

Cheers.

closed 2019-09-04 by EmmanuelP