AravisProject/aravis #255: arv-tool not finding USB-3 camera on Centos7

MarkRivers, 2019-06-18:
I have a Point Grey Grasshopper3 USB-3 camera connected to my Centos7 system. arv-tool-0.6 (aravis 0.6.3) is not finding the camera.

This is the output of lsusb:
```
[epics@TahoeC7 ~]$ lsusb
Bus 002 Device 004: ID 0bda:0328 Realtek Semiconductor Corp.
Bus 002 Device 006: ID 1e10:3300 Point Grey Research, Inc.
Bus 002 Device 002: ID 0424:5537 Standard Microsystems Corp.
Bus 002 Device 005: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 413c:2113 Dell Computer Corp.
Bus 001 Device 003: ID 413c:301a Dell Computer Corp.
Bus 001 Device 004: ID 04d8:0b29 Microchip Technology, Inc.
Bus 001 Device 002: ID 0424:2137 Standard Microsystems Corp.
Bus 001 Device 006: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
So the camera is on bus 002 device 006.

This shows the permissions on that device.
```
[epics@TahoeC7 ~]$ ls -l /dev/bus/usb/002
total 0
crw-rw-r-- 1 root root 189, 128 Jun 17 22:43 001
crw-rw-r-- 1 root root 189, 129 Jun 17 22:43 002
crw-rw-r-- 1 root root 189, 131 Jun 17 22:43 004
crw-rw-r-- 1 root root 189, 132 Jun 17 22:43 005
crw-rw-rw- 1 root root 189, 133 Jun 17 18:30 006
```
It has rw for everyone, which is how the udev aravis.rules configures it.

The problem is that arv-tool-0.6 does not find the camera:
```
[epics@TahoeC7 ~]$ arv-tool-0.6
Allied Vision Technologies-GC1380H (02-2142A)-02-2142A-06178 (164.54.160.58)
Allied Vision Technologies-GT1380 (02-2604A)-02-2604A-07008 (164.54.160.104)
Allied Vision Technologies-Manta G-507C (E0022705)-50-0503419258 (164.54.160.21)
Allied Vision Technologies-Manta_G-146C (E0020011)-50-0503317598 (164.54.160.62)
Point Grey Research-Blackfly BFLY-PGE-20E4C-13481965 (164.54.160.114)
```

That Point Grey camera that it does find is a GigE Blackfly, it is not the USB Grasshopper3.

Is there something I need to do to get it to find the camera?

This is a triple-boot system: Ubuntu 18, Centos 7, Windows 10.  The Grasshopper3 is working fine with aravis on Ubuntu 18, but it is not working on Centos 7.

MarkRivers, 2019-06-18:
I tried another Point Grey USB-3 camera in this same Centos7 system.  This is a BlackflyS 16S2M.  

It works fine with aravis on another Centos7 system.  

On this Centos7 system both cameras show up in lsusb and in /dev/bus/usb, but are not found by arv-tool.
```
[epics@TahoeC7 iocAravis]$ lsusb
Bus 002 Device 004: ID 0bda:0328 Realtek Semiconductor Corp.
Bus 002 Device 003: ID 1e10:3300 Point Grey Research, Inc.
Bus 002 Device 007: ID 0424:5537 Standard Microsystems Corp.
Bus 002 Device 005: ID 0bda:0411 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 413c:2113 Dell Computer Corp.
Bus 001 Device 016: ID 1e10:4000 Point Grey Research, Inc.
Bus 001 Device 015: ID 413c:301a Dell Computer Corp.
Bus 001 Device 010: ID 04d8:0b29 Microchip Technology, Inc.
Bus 001 Device 009: ID 0424:2137 Standard Microsystems Corp.
Bus 001 Device 006: ID 0bda:5411 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

[epics@TahoeC7 iocAravis]$ ls -l /dev/bus/usb/001/016
crw-rw-rw- 1 root root 189, 15 Jun 18 17:17 /dev/bus/usb/001/016
[epics@TahoeC7 iocAravis]$ ls -l /dev/bus/usb/002/003
crw-rw-rw-+ 1 root root 189, 130 Jun 18 13:20 /dev/bus/usb/002/003

[epics@TahoeC7 iocAravis]$ arv-tool-0.6
Allied Vision Technologies-GC1380H (02-2142A)-02-2142A-06178 (164.54.160.58)
Allied Vision Technologies-GT1380 (02-2604A)-02-2604A-07008 (164.54.160.104)
Allied Vision Technologies-Manta G-507C (E0022705)-50-0503419258 (164.54.160.21)
Allied Vision Technologies-Manta_G-146C (E0020011)-50-0503317598 (164.54.160.62)
FLIR-Oryx ORX-10G-51S5M-18011754 (192.168.0.2)
Point Grey Research-Blackfly BFLY-PGE-20E4C-13481965 (164.54.160.114)
```

EmmanuelP, 2019-09-04:
Hi Mark,

Did you check USB support is compiled on the system not finding you camera ?

The output of `arv-tool-0.6 -d all` should display USB related informations in its first line, like:

```
Found 1 USB3Vision device (among 18 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.0.4
...
```

EmmanuelP, 2020-08-19:
I'm closing this issue for now. Please reopen if you have the requested informations.

Cheers.

closed 2020-08-19 by EmmanuelP