AravisProject/aravis #258: PTP syncing support

mdgrogan, 2019-07-23:
I am trying to enable PTP syncing on a GigE camera. I haven't been able to find any information in the documentation or the header files. If anyone could offer some insight into this process, it would be greatly appreciated.

EmmanuelP, 2019-07-23:
Hi Matthew.

Which camera are you using ? What does the camera documentation say about PTP ?

mdgrogan, 2019-07-23:
I'm using an acA2500-14gm Basler ace GigE camera. PTP is supported, but disabled on startup. The product documentation outlines the procedure for enabling and checking the status of synchronization when using the pylon software suite.

EmmanuelP, 2019-07-24:
For what I have found  on the internet, it seems this functionality only requires to set the correct values of a set of Genicam features. It can be done programmatically using the arv_device_*feature_value family of functions. [https://aravisproject.github.io/docs/aravis-0.6/ArvDevice.html](https://aravisproject.github.io/docs/aravis-0.6/ArvDevice.html), or manually using arv-tool-0.6.

I'm closing. Please reposen this issue if you think a missing feature or a bug prevents you to configure the PTP support of your camera.

Cheers,

    Emmanuel.

closed 2019-07-24 by EmmanuelP