AravisProject/aravis #259: Opencv / Basler / imgproc

EdCha, 2019-07-24:
Hello

I am encountering a problem using my Basler Aviator 2300 with Opensuse 15.1
Using Aravis 6.3, Opencv 4.1 from source & Qarv (branch called Edouard, a slider issue with the Aviator fixed by Jure) 

Until now, where I had my acquisition machine running on Opensuse 42.3, there was no issue.
Time to move to a more up to date distribution. Opensuse 15.1 from scratch not an update

I am getting a segfault when displaying acquisition using Qarv.
**[ 8790.213388] QArv Cooker[26472]: segfault at 7efd9056c084 ip 00007efd880dd450 sp 00007efd64859380 error 4 in libopencv_imgproc.so.4.0.0[7efd87f0a000+1449000]**

I honestly have no idea, not being a developer. Aravis ? Qarv ? opencv ? OpenSuse ? libav ?

Jure has kindly spent a lot of time to get this branch  (Edouard / Aviator) working and he also has tested another Basler camera and reported Qarv functioning properly.
All other Qarv functions are operative and stable.

Arv-viewer is displaying the video, however does not offer the RGB12 option
Pylon seems to be working properly as well, although some warnings here and there, but still operative.

I am starting to wonder if my Aviator is not the source of the problem
Is there anyone out there who'd know about this error ? Any advice ?

Many thanks
Ed

a-andre, 2019-07-24:
Are you using OpenCV 4.1 or 4.0? At the top you say 4.1, but the error says 4.0.0. You might have linked against OpenCV 4.1 but for execution version 4.0 is loaded.

EmmanuelP, 2019-07-24:
Please attach a function call backtrace using gdb. In order to get a backtrace, launch your application with gdb:

```
gdb --args your_app your_parameters
```

Then type `run` in the gdb shell. Once the application has crashed, get the backtrace using: `bt full`.

Since it is most likely an aravis issue, I'm closing this issue. I'll reopen it if the backtrace says the contrary.

closed 2019-07-24 by EmmanuelP

EdCha, 2019-07-24:
> Are you using OpenCV 4.1 or 4.0? At the top you say 4.1, but the error says 4.0.0. You might have linked against OpenCV 4.1 but for execution version 4.0 is loaded.

Overlooked that, I tried both opencv and both are giving the same error, I copied the message from 4.0.0
E

EdCha, 2019-07-24:
Hello Emmanuel
My apologies if I post at the wrong place. Fairly new to this.
Backtrace is sending 
(gdb) bt full
#0  0x00007fffed71c1d0 in cv::Bayer2RGB_Invoker<unsigned char, cv::SIMDBayerInterpolator_8u>::operator()(cv::Range const&) const ()
    at /usr/local/lib64/libopencv_imgproc.so.4.1
#1  0x00007fffec514235 in (anonymous namespace)::ParallelLoopBodyWrapper::operator()(cv::Range const&) const ()
    at /usr/local/lib64/libopencv_core.so.4.1
#2  0x00007fffec5150a9 in cv::ParallelJob::execute(bool) () at /usr/local/lib64/libopencv_core.so.4.1
#3  0x00007fffec5179f9 in cv::WorkerThread::thread_body() () at /usr/local/lib64/libopencv_core.so.4.1
#4  0x00007fffec517bd9 in cv::WorkerThread::thread_loop_wrapper(void*) () at /usr/local/lib64/libopencv_core.so.4.1
#5  0x00007fffeb728569 in start_thread () at /lib64/libpthread.so.0
#6  0x00007fffeb45f9ef in clone () at /lib64/libc.so.6

EmmanuelP, 2019-07-25:
> My apologies if I post at the wrong place. 

No problem.

The 2 backtraces are different, the first crash happened in OpenCV, while the second happened in qarv. In any cas, it confirm it is most likely not an aravis issue.

Cheers.