AravisProject/aravis #260: Reported Min size

EdCha, 2019-07-24:
Hello

Using the latest version of Aravis from source (6.3 and master) and I get the minimum values for X/Y that I can grab as **4** and **444** from my Aviator 2300
May or may not be an issue, I prefer to report as it seems odd
Ed

EmmanuelP, 2019-07-24:
Hi EdCha,

This report doesn't have enough information to be useful. At least it should says what you expect. I'm closing it. Please reopen if you have additional informations.

Cheers,

    Emmanuel.

EdCha, 2019-08-01:
HI Emmanuel

Here is the situation, trying to, for say, have a small frame of 128 x128

arv-tool-0.6 control Width=128 Height=128
Basler-avA2300-25gc-21460462 (192.168.2.3)
Width = 128 (min:4;max:2356)
Height = 128 (min:**444**;max:1772)

Arv-viewer then display a frame of 128 x 128 but advertise a region of 128 x 444
If any of these values are modified to a number other than a multiple of 4 the returned frame is black.

Also camera specs max size are 2330 x1750 which differs from values found by Aravis

Please let me know if I can test anything to provide more relevant piece of information

Cheers
E

EmmanuelP, 2019-08-02:
Did you try your camera using Pylon ?

I would not worry about the maximum size, as sometimes the sensor has more pixels than what the camera manufacturer says.

Could you attach the output of `arv-tool-0.6 control Width=128 Height=128 --debug=all:3` ?

EdCha, 2019-08-05:
Hello Emmanuel

Pylon is detecting a min size of 128 x 444
Attached is the result of the debug command 
[Basler_report.txt](https://github.com/AravisProject/aravis/files/3465825/Basler_report.txt)

Thanks
E

EmmanuelP, 2019-08-05:
Could you also attach the genicam data, using `arv-tool-0.6 genicam`, please ?

EdCha, 2019-08-05:
There we go.. 
Thanks
E
[Genicam_basler.txt](https://github.com/AravisProject/aravis/files/3469413/Genicam_basler.txt)

EmmanuelP, 2019-08-06:
So, regarding the min/max values, they are simple register read for this device, no math involved. Everything looks fine.

But one thing aravis is doing wrong is the fact it doesn't check for min/max before writing a value. That was a deliberate choice, in order to be able to set a value even if the min/max reading was buggy. It also make things more efficient, as we don't have to get min/max value from the device, and caching does not work currently.

Now the genicam part is good enough, we could add the min/max check, but before we have to fix the register read/write cache. I'm opening a new issue about it.

Thanks for the report.

EmmanuelP, 2019-08-06:
For the record, relevant parts in log output and genicam data:

```sh

packet_type  = ack
command      = write-register-ack
size         = 4
id           = 65439
data index   =          1 (0x00000001)
00000000 00 00 00 83 00 04 ff 9f 00 00 00 01              ............    
packet_type  = cmd
command      = read-register-cmd
size         = 4
id           = 65440
address      =     197160 (0x00030228)
00000000 42 01 00 80 00 04 ff a0 00 03 02 28              B..........(    
packet_type  = ack
command      = read-register-ack
size         = 4
id           = 65440
success      =        444 (0x000001bc)
00000000 00 00 00 81 00 04 ff a0 00 00 01 bc              ............    
[GcRegisterNode::_get_integer_value] address = 0x30228, value = 0x1bc
packet_type  = cmd
command      = read-register-cmd
size         = 4
id           = 65441
address      =     197164 (0x0003022c)
00000000 42 01 00 80 00 04 ff a1 00 03 02 2c              B..........,    
packet_type  = ack
command      = read-register-ack
size         = 4
id           = 65441
success      =       1772 (0x000006ec)
00000000 00 00 00 81 00 04 ff a1 00 00 06 ec              ............    
[GcRegisterNode::_get_integer_value] address = 0x3022c, value = 0x6ec
packet_type  = cmd
command      = read-register-cmd
size         = 4
id           = 65442
address      =     197156 (0x00030224)
00000000 42 01 00 80 00 04 ff a2 00 03 02 24              B..........$    
packet_type  = ack
command      = read-register-ack
size         = 4
id           = 65442
success      =        128 (0x00000080)
00000000 00 00 00 81 00 04 ff a2 00 00 00 80              ............    
[GcRegisterNode::_get_integer_value] address = 0x30224, value = 0x80
Height = 128 (min:444;max:1772)
```

```xml
  <Integer Name="Height" NameSpace="Standard">
    <ToolTip>Sets the height of the area of interest in pixels</ToolTip>
    <Description>This value sets the height of the area of interest in pixels.</Description>
    <DisplayName>Height</DisplayName>
    <pIsImplemented>N350</pIsImplemented>
    <pIsAvailable>N351</pIsAvailable>
    <pIsLocked>N352</pIsLocked>
    <Streamable>Yes</Streamable>
    <pValue>N353</pValue>
    <pMin>N354</pMin>
    <pMax>N355</pMax>
    <pInc>N356</pInc>
    <Representation>Linear</Representation>
  </Integer>
  <IntReg Name="N354" NameSpace="Custom">
    <Address>0x30228</Address>
    <Length>4</Length>
    <AccessMode>RO</AccessMode>
    <pPort>Device</pPort>
    <pInvalidator>N685</pInvalidator>
    <pInvalidator>N29</pInvalidator>
    <pInvalidator>N3</pInvalidator>
    <pInvalidator>N396</pInvalidator>
    <pInvalidator>N382</pInvalidator>
    <Sign>Unsigned</Sign>
    <Endianess>BigEndian</Endianess>
  </IntReg>
```

closed 2019-08-06 by EmmanuelP