AravisProject/aravis #261: Support u3v camera with Payload Size Alignment = 0

gaoyichuan, 2019-08-01:
I have a Machine Vision brand camera, whose SI info register reads 0, thus Payload Size Alignment is 0, alignment = 2^0 == 1.
However in arvuvstream.c:234, which is:
`g_assert (alignment % 2 == 0 && alignment > 0);`
It wants the payload to be an even number. With that assertion removed, my camera works fine.
Also, according to the specification, this alignment value is totally fine.

So please consider removing this assertion, or change it according to the spec, thanks!

EmmanuelP, 2019-08-01:
Hi,

https://github.com/AravisProject/aravis/blame/9da9ce0a00399cfa51d7b84e48a87ac36ee527c4/src/arvuvstream.c#L231-L237

I guess the assertion can be safely removed. @JohannesBeck , could you comment on why you added this assertion ?

EmmanuelP, 2019-08-02:
Fix pushed to master. Thanks.

JohannesBeck, 2019-08-02:
Sorry for the late replay, but I think, the fix is not correct.
The SI info register of @gaoyichuan is zero, but the assert states `g_assert (alignment > 0);`. Also I realized, that the assert was not correct. The algorithm I used only works for alignment values of power of two, so that's what I tried to check with the assert. So I think the correct assert with the used algorithm is: `gassert(alignment & (alignment – 1) == 0 && alignment  > 0)`. The zero case needs to be handled differently. I created a pull request #263.

EmmanuelP, 2019-08-03:
SI_Info is 0, but the corresponding alignment variable is set to `1 << ((si_info & ARV_SI_INFO_ALIGNMENT_MASK) >> ARV_SI_INFO_ALIGNMENT_SHIFT);`, which will be 1 in this case.

I would not worry to find the exact assertion. May be the good approach would be to use SI_info properly masked and shifted as the input of the align function, and do the bit 1 shifting inside align.

JohannesBeck, 2019-08-03:
I missed the calculation of the alignment based on the SI_info register. I updated the PR to just fix the assert.

commit f216a4f2 (2019-08-04): uvstream: remove parity assertion on alignment

Fixes #261
closed 2019-08-03 by EmmanuelP