AravisProject/aravis #264: OSX build fail and Mercury camera

pavan4, 2019-08-03:
Hello,

When I try to build the project on Mac  I get the following error - 

```
Run-time dependency gobject-2.0 found: NO (tried cmake and framework)
meson.build:23:0: ERROR: Could not generate cargs for gobject-2.0: 
```

Could anyone point me in the right direction to build this project? I am a noob using machine vision cameras, apologies if this is a basic fix.

Also, I have a mercury galaxy viewer machine vision camera (Geni  compliant GigE) Have they been tested with Aravis?

EmmanuelP, 2019-08-03:
Hi,

Well, I'm also interested in any insight about this issue. The current Continuous Integration pipeline fails on OSX for the very same reason.

https://travis-ci.org/AravisProject/aravis/jobs/567093052#L2126

EmmanuelP, 2020-04-21:
I have just tried to make travis ci build reliably, but I'm giving up. It fails randomly and as I don't have an OSX system at hand, I can not debug the OSX build.

https://github.com/AravisProject/aravis/commit/feadecc47aac1083fb5bc258c6637273d3abc034

That being said, the commands in the travis file at least pass the meson configuration step. So please have a look at this file: https://github.com/AravisProject/aravis/blob/master/.travis.yml

> Also, I have a mercury galaxy viewer machine vision camera (Geni compliant GigE) Have they been tested with Aravis?

Not by me for sure. You should ask on aravis discourse.

Don't hesitate to reopen this issue if you think there is something we can do in aravis itself.

closed 2020-04-21 by EmmanuelP