AravisProject/aravis #278: Fake GV camera FireTestPacket support

EmmanuelP, 2019-08-15:
@highperformancecoder wrote:

I had a bit of a look at the packet size auto negotiation with eBUSPlayer (free windows download, or a paid Linux version) with wireshark. It seems it is writing a series of different packet sizes into the Stream Channel #0 Packet Size register, with a bit set to fire a test packet, and request no fragmentation. See attached image. But Aravis accepts the write register command, but because it doesn't know about that register, the value is ignored, same with the flags, so no test packet is fired. Eventually, eBUSPlayer gives up, and reports setting the packet size to the fallback value of 1476, although curiously in the trace it sets it to 2000 (which causes fragmentation on a standard MTU of 1500).

[image]

NB the only way I have been able to get aravis fake GV camera to work with eBUSPlayer is to clamp the value of data_size on line 363 of arvgvfakecamera.c to 16000. This value was determined empirically, I have no theory as to why it should be that. I'm not even sure what the variable gv_packet_size is supposed to represent. Jumbo packets would be fine with that value, but the maximum sized UDP packet is in theory 2^16, so this limitation reduces the effectiveness of loopback device connection (which eBUSPlayer refuses to recognise anyway).

EmmanuelP, 2019-08-17:
> NB the only way I have been able to get aravis fake GV camera to work with eBUSPlayer is to clamp the value of data_size on line 363 of arvgvfakecamera.c to 16000. This value was determined [...]

That should be fixed by b915fd1bd644436f40d53d4ad549fe668737725d

highperformancecoder, 2019-08-19:
On Sat, Aug 17, 2019 at 11:21:37AM -0700, Emmanuel Pacaud wrote:
> NB the only way I have been able to get aravis fake GV camera to work with
>     eBUSPlayer is to clamp the value of data_size on line 363 of
>     arvgvfakecamera.c to 16000. This value was determined [...]

Thank you. Yes I can verify that Aravis top of tree now works with eBUSPlayer.

Cheers

-- 

----------------------------------------------------------------------------
Dr Russell Standish                    Phone 0425 253119 (mobile)
Principal, High Performance Coders
Visiting Senior Research Fellow        hpcoder@hpcoders.com.au
Economics, Kingston University         http://www.hpcoders.com.au
----------------------------------------------------------------------------

closed 2025-02-16 by EmmanuelP