AravisProject/aravis #285: device not found

vahidVision, 2019-08-31:
hi
 i have usb3vision camera (dahua  A5501cu60) i installed Aravis and update aravis.rules by lsusb output but when i execute ./arv-device-test on my arm embedded platform with ubuntu
unfortunately it said device not found!!!

EmmanuelP, 2019-09-01:
What is the output of `arv-tool-0.8 --debug=all` and `sudo lsusb -v` ?

vahidVision, 2019-09-01:
thank you for your attention
i do following process 
1.download aravis-0.6.3.tar.xz
2.configure it successfully
3.make 
4.make install 
5.ldconfig
6. root@tegra-ubuntu:/home/ubuntu# lsusb -v
Bus 004 Device 004: ID 2e03:0001  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x2e03 
  idProduct          0x0001 
  bcdDevice           11.06
  iManufacturer           1 Dahua Technology
  iProduct                2 A5501CU60
  iSerial                 3 4H05114PAKCE6F3
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          116
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              224mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0 
      iFunction               4 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0 
      iInterface              5 Dahua Technology
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 06 01 02 07 08 09 0b 0a 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      1 
      iInterface             11 4H05114PAKCE6F3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      2 
      iInterface             12 7X3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           22
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
Device Status:     0x0001
  Self Powered

7.update aravis.rules by
# Daua
SUBSYSTEM=="usb", ATTRS{idVendor}=="2E03", MODE:="0666", TAG+="uaccess", TAG+="udev-acl"
# Basler
SUBSYSTEM=="usb", ATTRS{idVendor}=="2676", MODE:="0666", TAG+="uaccess", TAG+="udev-acl"
# The Imaging Source
SUBSYSTEM=="usb", ATTRS{idVendor}=="199e", MODE:="0666", TAG+="uaccess", TAG+="udev-acl"
# Point Grey Research
SUBSYSTEM=="usb", ATTRS{idVendor}=="1e10", MODE:="0666", TAG+="uaccess", TAG+="udev-acl"

8.copy aravis.rules to /etc/udev/rules.d

now where can i find arv-tool-0.8 to execute ?
thanks

EmmanuelP, 2019-09-01:
> now where can i find arv-tool-0.8 to execute ?

Oh sorry, it is arv-tool-0.6, which should be in `/usr/local/bin`, or `/usr/bin`.

EmmanuelP, 2019-09-01:
Did you try to run arv-tool as root ?

vahidVision, 2019-09-01:
yes 
1.sudo su
2.cd /usr/local/bin/
3.arv-tool-0.6
device not found
there is a another file in this directory so!!! 
arv-fake-gv-camera-0.6

EmmanuelP, 2019-09-01:
What is the output of `arv-tool-0.6 --debug=all:3` ?

vahidVision, 2019-09-01:
root@tegra-ubuntu:/usr/local/bin# arv-tool-0.6 --debug=all:3
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 192.168.1.166
No device found

EmmanuelP, 2019-09-01:
USB3 support is not compiled. Make sure you configure the compilation with `--enable-usb`. You will probably need to install libusb development package.

vahidVision, 2019-09-01:
thank you very much
 it seems to be OK
**1.** by installing libusb and  ./configure --enable-usb
  i get this result at end :
 Configuration:

  Compiler:                     gcc
  Compiler flags:               -pthread -I/usr/include/glib-2.0 -I/  usr/lib/arm-linux-gnueabihf/glib-2.0/include -I/usr/include/libxml2 -I/usr/include/libusb-1.0  
  Linker flags:                 -lm -pthread -lgio-2.0 -lgobject-2.0 -lxml2 -lgthread-2.0 -lglib-2.0 -lz -lusb-1.0  

  Build USB support:            yes
  Build packet socket support:  no
  Build viewer:                 no
  Build gstreamer plugin:       yes
  Build gstreamer-0.10 plugin:  no
  Build with fast heartbeat:    no
  Build with address sanatizer: no

**2.**root@tegra-ubuntu:/usr/local/bin#** arv-tool-0.6 
 Dahua Technology-4H05114PAKCE6F3 (USB3)

**3.** by run ./arv-device-test
resualt is:
Looking for the first available camera
vendor        = Dahua Technology
model         = A5501CU60

(lt-arv-device-test:25138): Aravis-CRITICAL **: arv_gc_string_get_value: assertion 'ARV_IS_GC_STRING (gc_string)' failed
device id     = (null)
sensor width  = 2592
sensor height = 2048
image width   = 2592 (max:2592)
image height  = 2048 (max:2048)
exposure            = 5000 (min:-1.79769e+308 - max:1.79769e+308)
exposure auto mode  = Off

(lt-arv-device-test:25138): GLib-GObject-WARNING **: invalid cast from 'ArvGcFloatNode' to 'ArvGcInteger'

(lt-arv-device-test:25138): Aravis-CRITICAL **: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

(lt-arv-device-test:25138): GLib-GObject-WARNING **: invalid cast from 'ArvGcFloatNode' to 'ArvGcInteger'

(lt-arv-device-test:25138): Aravis-CRITICAL **: arv_gc_integer_get_max: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed

(lt-arv-device-test:25138): GLib-GObject-WARNING **: invalid cast from 'ArvGcFloatNode' to 'ArvGcInteger'

(lt-arv-device-test:25138): Aravis-CRITICAL **: arv_gc_integer_get_min: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
gain                = 0 (min:0 - max:0)
gain auto mode      = Off
trigger selector    = FrameStart
reverse x           = FALSE
**
Aravis:ERROR:arvuvstream.c:234:align: assertion failed: (alignment % 2 == 0 && alignment > 0)
Aborted

it seems ./arv-device-test still not work properly
by such this assertion failed can i develop my own program by this camera?

EmmanuelP, 2019-09-01:
arv-test is an unfinished test program. Please don't use it, but tests/arv-camera-test instead.

The assertion test should be fixed in the 0.6 branch, but there is no release yet. I'm going make a 0.6.4 release today. Meanwhile, you can try to just remove the assertion.

EmmanuelP, 2019-09-01:
New release done:

https://aravis-project.discourse.group/t/aravis-0-6-4-stable-release/18?u=emmanuel

vahidVision, 2019-09-01:
by your new version (0-6-4)everything become OK
arv-camera-test run without any bug report
thank you

EmmanuelP, 2019-09-02:
Ok, Thanks.

closed 2019-09-02 by EmmanuelP