AravisProject/aravis #287: opencv combination with aravis

vahidVision, 2019-09-02:
hi 
according to this https://github.com/kushalvyas/Aravis-OpenCV-Wrapper/blob/master/src/Camera.cpp

i try to render video real time on opencv window
i try following code by adding only        std::cout<<buffer_size<<std::endl to above code

IplImage* Camera::readFrameIPL(){
    ArvBuffer* _buffer;
    IplImage* src;
    size_t buffer_size;

    // get buffer data
    _buffer = arv_stream_pop_buffer(stream);
    if (_buffer != NULL){
        // copy buffer data
        framebuffer = (void*)arv_buffer_get_data(_buffer, &buffer_size);
        std::cout<<buffer_size<<std::endl;/// i add this line
        arv_stream_push_buffer(stream, _buffer);
        // create IplImage
        src = cvCreateImage( cvSize(1280, 960), IPL_DEPTH_8U, 1);
        cvInitImageHeader(src, cvSize(1280, 960), IPL_DEPTH_8U, 1, IPL_ORIGIN_TL, 4);
        cvSetData(src, framebuffer, src->widthStep);

        return src;
    }
    return NULL;
}

and on main use this
 while(!kbhit()){
        IplImage* image;
        image = cam.readFrameIPL();
        if(image){		
		 cv::Mat matimage = cv::cvarrToMat(image,false);
		imshow("video",matimage );
		}
		else
			 std::cout<<"error to read frame";
		usleep(5000); 
    
    }  
    cam.disconnect();	 

i achieved following result: 

1. on    std::cout<<buffer_size<<std::endl in Camera::readFrameIPL() method
print  5308416 continuously (2592*2048)
that seems it is GreyScale because  RGB should be(2592*2048*3) not  2592*2048 =5308416
why?
2.on Opencv window the full noisy image without any recognizable video content is shown continuously !!!
thanks

vahidVision, 2019-09-02:
i mean 2592**2048 and 2592**2048**3

vahidVision, 2019-09-02:
multiply (2592) and (2048) width an height of image

EmmanuelP, 2019-09-03:
Hi,

Please don't ask about aravis usage using github issue board.

I have recently created a discourse instance for Aravis discussions here: https://aravis-project.discourse.group/

You should be able to post your question here using your github credentials.

Thanks.

closed 2019-09-03 by EmmanuelP

vahidVision, 2019-09-03:
OK thanks