AravisProject/aravis #288: arv-viewer shows blank and black screen

vahidVision, 2019-09-03:
hi 
when i configure Aravis with arv-viewer and run it as superuser 
it recognize my camera and i select it but when i play video i get nothing except black empty screen
!!!

vahidVision, 2019-09-03:
my configure result is this
  Build USB support:            yes
  Build packet socket support:  no
  Build viewer:                 yes
  Build gstreamer plugin:       yes
  Build gstreamer-0.10 plugin:  no
  Build with fast heartbeat:    no
  Build with address sanitizer: no

EmmanuelP, 2019-09-03:
Please try `./tests/arv-camera-test -d all`

vahidVision, 2019-09-06:
everything is ok and cause of black screen was the exposure time that should be increase 
but now i have another problem by arv-viewer , the video has been shown with about 4 second delay i dont really know why?
should i change any setting too to capture real time video without any delay
i run arv-viewer on ubuntu 16.04 in virtual machine on my windows PC , is it the possible reason for this delay?

vahidVision, 2019-09-06:
excuse me 
 when i execute arv-viewer again, delay time become too small !!!!
i think it is good now 
thank you very much for your attention

closed 2019-09-06 by EmmanuelP