AravisProject/aravis #289: Chunk data does not work with Lucid Vision camera

NathanMOlson, 2019-09-03:
**Describe the bug**
Aravis fails when using chunk data with Lucid Vision camera.  From Lucid Vision tech support:

"I think the issue is Aravis does not yet successfully parse our chunk data implementation. We use the newer "image extended chunk data" implementation.

In src/arvbuffer.c in the arv_buffer_get_chunk_data function, there are checks for:
ARV_BUFFER_PAYLOAD_TYPE_CHUNK_DATA, which corresponds to the older chunk data implementation (0x0004, tagged blocks of data)
ARV_BUFFER_PAYLOAD_TYPE_EXTENDED_CHUNK_DATA, which corresponds to a different extended chunk data implementation (0x0005, also tagged blocks of data but also deprecated)

Our implementation is "image extended chunk data" which is the recommendation for newer GigE Vision 2.0 designs. In this implementation, each data payload is considered a chunk including image data payload which would be the first chunk. The image data payload would be followed by the other metadata payload chunks, e.x. ExposureTime and Gain."

**To Reproduce**
Compile and run this code:

```
#include "arv.h"
#include <iostream>
#include <stdint.h>
#include <vector>

using namespace std;

int main()
{
    ArvInterface interface;
    arv_update_device_list();
    uint32_t n_devices = arv_get_n_devices();

    if (n_devices == 0)
    {
        cout<<"No GenICam cameras available"<<endl;
        return(-1);
    }
    ArvDevice* p_device;
    ArvCamera* p_cam = arv_camera_new(arv_get_device_id(0));
    p_device = arv_camera_get_device(p_cam);

    arv_camera_stop_acquisition(p_cam);

    cout<<"Device vendor: "<<arv_camera_get_vendor_name(p_cam)<<endl;
    cout<<"Device model: "<<arv_camera_get_model_name(p_cam)<<endl;

    //cout<<"Device family: "<<arv_device_get_string_feature_value(p_device, "DeviceFamilyName")<<endl;
    cout<<"Device version: "<<arv_device_get_string_feature_value(p_device, "DeviceVersion")<<endl;
    //cout<<"Firmware version: "<<arv_device_get_string_feature_value(p_device, "DeviceFirmwareVersion")<<endl;
    //cout<<"Serial number: "<<arv_device_get_string_feature_value(p_device, "DeviceSerialNumber")<<endl;
    //cout<<"Sensor Description: "<<arv_device_get_string_feature_value(p_device, "SensorDescription")<<endl;

    cout<<"Aravis library "<<ARAVIS_MAJOR_VERSION<<"."<<ARAVIS_MINOR_VERSION<<"."<<ARAVIS_MICRO_VERSION<<endl;

    arv_device_set_string_feature_value (p_device, "TriggerSelector", "AcquisitionStart");
    arv_device_set_string_feature_value (p_device, "TriggerMode", "Off");
    arv_device_set_string_feature_value (p_device, "TriggerSelector", "FrameStart");
    arv_device_set_string_feature_value (p_device, "TriggerActivation", "RisingEdge");
    arv_device_set_string_feature_value (p_device, "TriggerSource", "Software");
    arv_device_set_string_feature_value (p_device, "TriggerMode", "On");

    uint32_t n_formats;
    int64_t* formats = arv_camera_get_available_pixel_formats(p_cam, &n_formats);
    vector<uint32_t> desired_formats;
    desired_formats.push_back(ARV_PIXEL_FORMAT_BAYER_RG_12_PACKED);
    desired_formats.push_back(ARV_PIXEL_FORMAT_BAYER_RG_12);
    desired_formats.push_back(ARV_PIXEL_FORMAT_BAYER_RG_16);
    desired_formats.push_back(ARV_PIXEL_FORMAT_MONO_12_PACKED);
    desired_formats.push_back(ARV_PIXEL_FORMAT_MONO_12);
    desired_formats.push_back(ARV_PIXEL_FORMAT_MONO_16);
    bool format_set = false;
    for(int i = 0; i < desired_formats.size(); i++)
    {
        for(int j = 0; j < n_formats; j++)
        {
            if(formats[j] == desired_formats[i])
            {
                arv_camera_set_pixel_format(p_cam, desired_formats[i]);
                format_set = true;
                break;
            }
        }
        if(format_set)
        {
            break;
        }
    }
    if(!format_set)
    {
        cout<<"Failed to set pixel format, available pixel formats are:";
        for(int j = 0; j < n_formats; j++)
        {
            cout<<"\t"<<formats[j];
        }
        cout<<endl;
    }

    cout<<"Pixel format: "<<arv_camera_get_pixel_format_as_string(p_cam)<<endl;

    //Enable embedded info
    arv_camera_set_chunks(p_cam, "ExposureTime Gain");
    ArvChunkParser* p_chunk_parser = arv_camera_create_chunk_parser(p_cam);

    arv_device_set_string_feature_value (p_device, "BalanceWhiteAuto", "Off");
    arv_device_set_string_feature_value (p_device, "BalanceRatioSelector", "Blue");
    arv_device_set_float_feature_value (p_device, "BalanceRatio", 1.633f);
    arv_device_set_string_feature_value (p_device, "BalanceRatioSelector", "Red");
    arv_device_set_float_feature_value (p_device, "BalanceRatio", 1.716f);

    arv_device_set_float_feature_value(p_device, "BlackLevel", 0.);
    arv_device_set_boolean_feature_value(p_device, "IspEnable", false);
    arv_device_set_boolean_feature_value(p_device, "GammaEnable", false);
    arv_device_set_string_feature_value(p_device, "AdcBitDepth", "Bit12");

    int32_t width, height;
    arv_camera_get_sensor_size(p_cam, &width, &height);
    cout<<"Sensor size: "<<width<<"x"<<height<<endl;

    if(arv_camera_is_gv_device(p_cam))
    {
        arv_camera_gv_set_packet_size(p_cam, 9000);
        arv_camera_gv_set_packet_delay(p_cam, 90000);
    }

    int32_t payload = arv_camera_get_payload(p_cam);

    ArvStream* p_stream = arv_camera_create_stream(p_cam, NULL, NULL);
    arv_stream_push_buffer(p_stream, arv_buffer_new(payload, NULL));
    arv_stream_push_buffer(p_stream, arv_buffer_new(payload, NULL));
    arv_camera_start_acquisition(p_cam);

    arv_camera_set_gain_auto(p_cam, ARV_AUTO_OFF);
    arv_camera_set_exposure_time_auto(p_cam, ARV_AUTO_OFF);

    arv_device_set_float_feature_value(p_device, "AutoExposureTimeUpperLimit", 5000);
    arv_device_set_float_feature_value(p_device, "AutoExposureExposureTimeUpperLimit", 5000);

    arv_camera_software_trigger(p_cam);

    ArvBuffer* p_buf = arv_stream_pop_buffer(p_stream);
    ArvBufferStatus status = arv_buffer_get_status(p_buf);
    if(status == ARV_BUFFER_STATUS_SUCCESS)
    {
        ArvPixelFormat pixel_format = arv_buffer_get_image_pixel_format(p_buf);
        cout<<"ChunkExposureTime: "<<arv_chunk_parser_get_float_value(p_chunk_parser, p_buf, "ChunkExposureTime")<<endl;
        cout<<"ChunkGain: "<<arv_chunk_parser_get_float_value(p_chunk_parser, p_buf, "ChunkGain")<<endl;
    }
    arv_stream_push_buffer(p_stream, p_buf);

    arv_camera_stop_acquisition(p_cam);
	g_object_unref(p_stream);
	g_object_unref(p_cam);
    return 0;
}
```

**Camera description:**
 - Manufacturer Lucid Vision
 - Model PHX120S
 - Interface: GigE

**Platform description:**
 - Aravis version 0.6.3
 - OS: Ubuntu 16.04
 - Hardware: ARM64 (Jetson TX2)

EmmanuelP, 2019-09-04:
Hi Nathan,

Thanks for the detailed report. What would help is a wireshark snapshot of the network traffic during the execution of your sample program. Could you try to capture it and attach the file here.

In order to keep the capture file to a reasonable size, only capture the GVCP and GVSP packets, and ignore the camera initialization which happens during arv_camera_new.

Thanks.

NathanMOlson, 2019-09-04:
Here's a pcap with all GVCP, GVSP, and UDP packets from the software trigger to the end of the program.

[arvcam_starting_at_trigger.zip](https://github.com/AravisProject/aravis/files/3576005/arvcam_starting_at_trigger.zip)

EmmanuelP, 2019-09-05:
Hi Nathan,

I don't see GVSP packets in the pcap file.

NathanMOlson, 2019-09-05:
Me neither... Here's a complete pcap file from a run of the program.  There are no GVSP packets.

[arvcam.zip](https://github.com/AravisProject/aravis/files/3579953/arvcam.zip)

NathanMOlson, 2019-09-11:
Whoops, I was on a different system and the MTU was not set to allow jumbo frames.  Here's a pcap with GVSP packets.  I switched the ROI to 32x32 to reduce file size.

[arvcam8.zip](https://github.com/AravisProject/aravis/files/3598426/arvcam8.zip)

EmmanuelP, 2019-09-11:
Hi Nathan,

Hopefully, I will receive a Lucid camera, thanks to LucidVision. That will makes the development of extended chunk support way easier.

NathanMOlson, 2019-09-12:
Here's a patch that seems to work (applied to v0.6.2).  I am surprised that it does so, I did not have to change anything related to parsing the chunks.  I basically just added a new packet type to the list and the rest took care of itself.  I am receiving ExposureTime and Gain chunks.
[lucid.txt](https://github.com/AravisProject/aravis/files/3603428/lucid.txt)

One oddity I noticed is that the exposure time units are strange.  There are two feature nodes, ExposureTime (float) and ExposureTimeRaw (integer).  I expect these both tin microseconds, but when I read them with Aravis they are different by a factor of 7.1362e44.  Probably a question for LucidVision, but I thought I'd mention it.

EmmanuelP, 2019-09-18:
Hi Nathan,

I have created a new branch with your patch: https://github.com/AravisProject/aravis/tree/wip/extended-chunk

Regarding Exposure values, is it when you try to read chunk data, or are they just normal feature values ?

Could you try to extract the relevant part in the genicam xml data you can get using `arv-tool-0.6 genicam` ?

NathanMOlson, 2019-09-26:
The problem with ExposureTime affected both chunk data and the feature value.  The problem affected Lucid Vision PHX120S-CC with firmware 1.5.0.0.  The problem is fixed in firmware 1.12.0.0.

EmmanuelP, 2019-10-05:
@NathanMOlson It looks like ImageExtendedChunk payload is the same as ChunkData payload, so your patch is fine. I'm going to push it to master. Could you send me your email address ? I'd like to push the changes with you as the author.

EmmanuelP, 2019-10-05:
Thanks Nathan. Pushed to master.

closed 2019-10-05 by EmmanuelP

Nostmr, 2023-09-07:
@EmmanuelP I have the same problem with my PHX023S-C camera. How do I solve it ?

Nostmr, 2023-09-07:
Create Device Failed:

TimeoutException (GC_ERR_TIMEOUT): Unable to open device (System.cpp, Arena::System::CreateDevice; GenTL::IFOpenDevice); Error Stack: -1011 GenTL::GC_ERR_TIMEOUT(-1011) : GenTL::IFOpenDevice():2157: GenTL::GC_ERR_TIMEOUT(-1011) -1011 GenTL::GC_ERR_TIMEOUT(-1011) : GenTL::InterfaceGev::OpenDevice():566: GenTL::GC_ERR_TIMEOUT(-1011) -1011 GenTL::GC_ERR_TIMEOUT(-1011) : GenTL::DeviceGev::Open():619: failed to read GVCP Capability (0x0934) register -1011 GenTL::GC_ERR_TIMEOUT(-1011) : GenTL::HALGev::ReadReg Ack is invalid: -1019 : (file 'System.cpp', line 682)