AravisProject/aravis #290: "Failed to claim USB interface to 'LightWise Allegro CMV4000 Mono - #00418002' " problem

dpwoo, 2019-09-10:
Hi ：
    I am working with a LW-AL-CMV4000M from ISG cameras.  I am using usb3vision and aravis-0.6.4 to operate this camera. And i meet some issue. The cmd-line program output like this:
   root@odroid:/home/odroid/spectral/aravis-ARAVIS_0_6_4/tests/.libs# ./arv-camera-test
   Looking for the first available camera
   [arvsystem.c 401] USB3Vision
   Found 1 USB3Vision device (among 10 USB devices)
   [UvDevice::new] Vendor  = Imaging Solutions Group
   [UvDevice::new] Product = LightWise Allegro CMV4000 Mono
   [UvDevice::new] S/N     = 00418002
   [UvDevice::new] Using control endpoint 1, interface 0
   [UvDevice::new] Using data endpoint 3, interface 1
   [UvDevice::new] [732, ret=-6]Failed to claim USB control interface to 'LightWise Allegro CMV4000 Mono - #00418002'
   LIBUSB_ERROR_BUSY

   When tracking the code， i found error occured in libusb1.0 ”int r = ioctl(fd, IOCTL_USBFS_CLAIMINTF, &iface);”
   and ioctl return error=EBUSY

  And my platform environment is：
   odroid@odroid:~$ lsb_release -a
   Distributor ID: Ubuntu
   Description:    Ubuntu 18.04.1 LTS
   Release:        18.04
   Codename:       bionic
   odroid@odroid:~$ uname -a
   Linux odroid 4.14.85-152 #1 SMP PREEMPT Mon Dec 3 03:00:02 -02 2018 armv7l armv7l armv7l GNU/Linux
   By the way，i now just using USB3.0 Micro-B connector provide power. 
   Can you provide any help with debugging this issue? 
   Thanks!

EmmanuelP, 2019-09-10:
Hi,

Did you create an entry for your camera in the udev file, like explained here: https://github.com/AravisProject/aravis#usb-permissions ?

The device can also be already controlled by a kernel module. Could you check the kernel logs when you are connecting the device ? You can use `sudo tail -f /var/log/messages` or `sudo journalctl -f` and watch for the output.

dpwoo, 2019-09-10:
Thank you for your reply！
1）Yes， i already copy aravis.rules to /etc/udev/rules.d. And its contents：
  SUBSYSTEM=="usb", ATTRS{idVendor}=="2978", MODE:="0666", TAG+="uaccess", TAG+="udev-acl"
2）I don't know if usb3vision dirver(u3v.ko) works welll. But when i insmod u3v.ko into kernel and rmmod u3v.ko from kernel. jornalctl -f output like this:
journalctl -f
 Sep 10 09:26:40 odroid kernel: usb 4-1.1: USB3 Vision Driver: Registering device 3
Sep 10 09:26:40 odroid kernel: usbcore: registered new interface driver u3v
Sep 10 09:26:40 odroid kernel: u3v: USB3 Vision Driver
Sep 10 09:26:40 odroid upowerd[958]: unhandled action 'bind' on /sys/devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.0
Sep 10 09:26:40 odroid upowerd[958]: unhandled action 'bind' on /sys/devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.1
Sep 10 09:27:01 odroid su[1286]: Successful su for root by odroid
Sep 10 09:27:01 odroid su[1286]: + /dev/pts/1 odroid:root
Sep 10 09:27:01 odroid su[1286]: pam_unix(su:session): session opened for user root by odroid(uid=1000)
Sep 10 09:27:01 odroid su[1286]: pam_systemd(su:session): Cannot create session: Already running in a session
Sep 10 09:28:15 odroid sudo[1302]:   odroid : TTY=pts/0 ; PWD=/home/odroid/install/bin ; USER=root ; COMMAND=/sbin/rmmod u3v
Sep 10 09:28:15 odroid sudo[1302]: pam_unix(sudo:session): session opened for user root by odroid(uid=0)
Sep 10 09:28:15 odroid kernel: usbcore: deregistering interface driver u3v
Sep 10 09:28:15 odroid kernel: usb 4-1.1: u3v_disconnect: U3V device removed
Sep 10 09:28:15 odroid upowerd[958]: unhandled action 'unbind' on /sys/devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.1
Sep 10 09:28:15 odroid upowerd[958]: unhandled action 'unbind' on /sys/devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.0
 It seems the driver not work corrrect?

 Thanks again!

EmmanuelP, 2019-09-10:
Aravis does not use u3v kernel driver, but libusb instead. Make sure u3v.ko is not loaded when you try to use Aravis.

dpwoo, 2019-09-10:
Thank you so much!
 I remove u3v.ko, and it works!!!

closed 2019-09-10 by EmmanuelP