AravisProject/aravis #295: [[UvDevice::read_memory] Unexpected answer (0x8fff) problem

dpwoo, 2019-09-24:
Hi，
I am using LW-AL-CMV4000M from ISG cameras and aravis-0.6.4. When i run arv-camera-test, it sometimes failed, and output like this, i attach the output log:
###################
packet_type  = cmd
command      = read-memory-cmd
size         = 12
id           = 65326
address      = 0x0000000000020008
size         =         64 (0x00000040)
00000000 55 33 56 43 00 40 00 08 0c 00 2e ff 08 00 02 00  U3VC.@..........
00000010 00 00 00 00 00 00 40 00                          ......@.        
packet_type  = (null)
command      = read-memory-ack
size         = 0
id           = 65326
00000000 55 33 56 43 ff 8f 01 08 00 00 2e ff              U3VC........    
[[UvDevice::read_memory] Unexpected answer (0x8fff)
[UvDevice::_bootstrap] Error during memory read
[UvDevice::new] Failed to bootstrap USB device
#####################################
Running environment：ubuntu 18.04， aarch64

Can you provide any help with debugging this issue?
Thanks

dpwoo, 2019-09-24:
Here is arv-camera-test complete output:

dpwoo, 2019-09-27:
This problem may be due to my incorrect operation. For example, in image acquisition, i kill the arv-camera-test, and then the problem arises. And i have to power off camera and then power up again to make camera nomal.

EmmanuelP, 2019-09-27:
Hi,

That is still interesting, may be there is something to do in aravis.

Could you try to get a debug output when the problem arises, by setting the ARV_DEBUG environment variable using `export ARV_DEBUG=all:3` ?

Thanks.

dpwoo, 2019-09-27:
Here is the debug output. 
[arv-camera-test.log](https://github.com/AravisProject/aravis/files/3663180/arv-camera-test.log)

When the problem arises, restart system didn't work. So I think maybe the camera is in a wrong state at this time. In the GenCp Version 1.2, the status code 0x8fff description as "Generic error".

EmmanuelP, 2019-10-10:
Thanks @dpwoo for the debug output.

The 'Generic error' is not really helpful. On a positive note, the fact the error happens after successful memory reads may suggest there is something to do that would put the camera back in a working state.

You may try to contact ISG support and see if they are willing to help us to fix this issue.