AravisProject/aravis #301: Problem while replugging the camera

crabsmaps, 2019-10-09:
**Describe the bug**
I have application that operates two basler USB cameras. This application handles re-plugging of the cameras. Most of the times when I re-plug the camera everything goes smoothly: aravis is being able to claim USB interface and operate the camera (configure the camera and gather images). But sometimes re-plugging of one of the cameras causes *both* cameras to be in a broken state.
I can access cameras in this state in a sense that I can query genicam data and different feature values, but it fails to return images, emitting various "USB transfer error" - LIBUSB_ERROR_TIMEOUT or LIBUSB_ERROR_IO

**To Reproduce**
Steps to reproduce the behavior: Unfortunately I cannot reproduce this issue reliably even with my own application. But basically it is: have two usb cameras running and acquiring images, unplug and re-plug one of the cameras, sometimes this causes *both* cameras to enter in a broken state in which they fail to get any images.

**Expected behavior**
I expected that re-plugging of one of the cameras should not affect the other camera.

**Camera description:**
 - Manufacturer 2x Basler
 - Model 2x acA4096-30uc
 - Interface 2x USB3

**Platform description:**
 - Aravis version 0.6.2
 - OS: Ubuntu 18.04
 - Hardware x86_64

**Additional context**
Here is the log of ARV_DEBUG=all:6 when I try to start cameras in already broken state.  [broken-state.log](https://github.com/AravisProject/aravis/files/3708006/broken-state.log)
I also have a larger log where this issue is triggered with libusb debug information, but I need to find a way to clean-up this log a bit since it is too big to upload and to read.

(Powercycling of the cameras of course helps, but I was hoping to avoid that.)

I didn't managed to pinpoint exact reason for that type of behavior but I have just a minor concern that might or might not contribute to this. Sometimes *during* the re-plug I'm getting LIBUSB_ERROR_PIPE error, and it seems like this kind of error requires a call to libusb_clear_halt before retrying libusb_bulk_transfer (as it is described [here](https://devzone.nordicsemi.com/f/nordic-q-a/41390/usb-endpoint-transfer-pipe-error))
>Looking at the documenation here, libusb_bulk_transfer () returning LIBUSB_ERROR_PIPE , means that the endpoint is halted. I think you might need to call libusb_clear_halt, and then try again.

crabsmaps, 2019-10-09:
If that helps, here is a log with LIBUSB_DEBUG=4 of trying to run a single camera in a broken state. 
[libusb-broken-state-single-camera.log](https://github.com/AravisProject/aravis/files/3708358/libusb-broken-state-single-camera.log)

crabsmaps, 2019-10-09:
Generally speaking this might be not even an issue of aravis, I'm just desperate to get some input on what might have gone wrong.

EmmanuelP, 2019-10-10:
Hi Dmitry,

I've tried to implement the suggested call to libusb_clear_halt. Please try this branch: https://github.com/AravisProject/aravis/tree/wip/libusb-epipe

Or, if it is too much work to port your code to aravis master, I think you should be able to apply the last commit to your copy of aravis.

crabsmaps, 2019-10-11:
Thanks for the proposed solution!

I tried both your branch and the patch on top of 0.6.2. From the documentation and from the examples of libusb your proposed solution totally makes sense, but unfortunately doesn't resolve the issue. Behavior is essentially stayed the same, after initial LIBUSB_ERROR_PIPE as expected libusb_clear_halt is called, but that only causes LIBUSB_ERROR_IO to appear on every call of libusb_bulk_transfer afterwards.

I also tried calling libusb_clear_halt on LIBUSB_ERROR_IO or LIBUSB_ERROR_PIPE - this does help recover in most cases, but
1. this usage on libusb_clear_halt is not justified on LIBUSB_ERROR_IO (there is no mention in documentation that one should do that)
2. this introduces another problem: again *sometimes* after calling libusb_clear_halt device will timeout every bulk transfer with LIBUSB_ERROR_TIMEOUT

===
I'll try to again rephrase and reinstate the problem if anybody encounters that as well.

Setup: two usb3 cameras (left and right) connected by hardware trigger. Left camera is a master camera, meaning that it is the one that issues trigger.

Issue: during live operation (while both cameras are grabbing frames) right camera gets replugged. Usually it doesn't cause any problems, but *sometimes* this causes left camera not to deliver any frames.

Some additional facts:
- left camera in a broken state still accessible via genicam interface
- left camera in a broken state still sends trigger signals over the trigger cable to the right camera
- arv_stream_timeout_pop_buffer "fails" on the left camera, returning NULL
- ARV_DEBUG=sp:6 prints over and over again error message
```
Asking for 1048576 bytes
USB transfer error: LIBUSB_ERROR_IO
```
- LIBUSB_DEBUG=4 print over and over again error message
```
[ 3.338241] [000027ac] libusb: debug [usbi_handle_transfer_completion] transfer 0x56047b8d7440 has callback 0x7f9457df2b30
[ 3.338245] [000027ac] libusb: debug [sync_transfer_cb] actual_length=16
[ 3.338255] [000027ac] libusb: debug [libusb_free_transfer] transfer 0x56047b8d7440
[ 3.338263] [000027d2] libusb: debug [libusb_free_transfer] transfer 0x7f9450102470
[ 3.338280] [000027d2] libusb: debug [libusb_alloc_transfer] transfer 0x7f9450102470
[ 3.338288] [000027d2] libusb: debug [libusb_submit_transfer] transfer 0x7f9450102470
[ 3.338296] [000027d2] libusb: debug [add_to_flying_list] arm timerfd for timeout in 50ms (first in line)
[ 3.338307] [000027d2] libusb: debug [submit_bulk_transfer] need 1 urbs for new transfer with length 1048576
[ 3.338347] [000027d2] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 3.338355] [000027d2] libusb: debug [handle_events] poll() 3 fds with timeout in 60000ms
[ 3.341673] [000027d2] libusb: debug [handle_events] poll() returned 1
[ 3.341704] [000027d2] libusb: debug [reap_for_handle] urb type=3 status=-71 transferred=0
[ 3.341713] [000027d2] libusb: debug [handle_bulk_completion] handling completion status -71 of bulk urb 1/1
[ 3.341722] [000027d2] libusb: debug [handle_bulk_completion] low level error -71
[ 3.341730] [000027d2] libusb: debug [disarm_timerfd] 
```

crabsmaps, 2019-10-11:
Tried opening camera in a broken state with basler's PylonViewerApp and it manages to get the images but with glitches (random blacked-out regions of the image) and it also prints out in a console
```
Unable to ensure stalled pipe 0, status=0xe2100001 LIBUSB_ERROR_IO
```

I guess this might be a device specific issue, if I don't get any more information by the next week I think I'll close this issue

crabsmaps, 2019-10-11:
Also interesting moment that number of LIBUSB_ERROR_IO is considerably small while running PylonViewerApp. It might be attributed that by default it is configured to use smaller usb packet size during transmission (in pylon it's 65536, and in aravis it's ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE=1048576)

EmmanuelP, 2019-10-15:
Did you try to play with the value of `ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE` ?

closed 2019-10-15 by EmmanuelP

crabsmaps, 2019-10-16:
Yes, I tried playing with this value, but this doesn't help.
I reproduced the issue on different machine with different cables. PylonViewer can somehow recover the camera from broken state.
I haven't figured out what is the reason for this behavior, but "fixed" it by issuing DeviceReset on several failed attempts to pop the buffer.
I imagine that this is not a real fix for anything.

commit 83c75573 (2019-10-22): uvdevice: reset data endpoint in _open_usb_device

Fixes #301

With this PR I no longer have problems with replugging the camera.

I think that the main reason for that problem was that the camera
has been in a halted state (see Figure 5-3:  Transmission States in
[USB3 Vision Specification](https://www.visiononline.org/userAssets/aiaUploads/file/USB3_Vision_Specification_v1-0-1.pdf))

In order to return camera back to the idle state I'm setting data
endpoint in a halted state and immediately clearing that state
(this solution I picked up from [here](https://github.com/ni/usb3vision/blob/maste [... cut]