AravisProject/aravis #302: Warnings when compiling docs

lovesegfault, 2019-10-17:
In packaging 0.7.2 for Gentoo I encountered the following
 
```
Building documentation for aravis
Unescaped left brace in regex is passed through in regex; marked by <-- HERE in m/(.*?){ <-- HERE / at /usr/bin/gtkdoc-scan line 735.

html/ArvXmlSchema.html:84: warning: no link for: 'GFile' -> (<span class="type">GFile</span>).
html/ArvXmlSchema.html:101: warning: no link for: 'GError' -> (<span class="type">GError</span>).
html/ArvEvaluator.html:224: warning: no link for: 'NULL:CAPS' -> (<code class="literal">NULL</code>).
html/ArvDomDocument.html:271: warning: no link for: 'GType' -> (<span class="type">GType</span>).
html/ArvDomDocument.html:359: warning: no link for: 'GOutputStream' -> (<span class="type">GOutputStream</span>).
html/ArvFakeCamera.html:305: warning: no link for: 'GInetAddress' -> (<span class="type">GInetAddress</span>).
html/ArvFakeCamera.html:355: warning: no link for: 'GSocketAddress' -> (<span class="type">GSocketAddress</span>).
html/ArvGvFakeCamera.html:250: warning: no link for: 'TRUE:CAPS' -> (<code class="literal">TRUE</code>).
html/ArvBuffer.html:571: warning: no link for: 'GDestroyNotify' -> (<span class="type">GDestroyNotify</span>).
html/ArvStream.html:177: warning: no link for: 'G-SIGNAL-RUN-LAST:CAPS' -> (Run Last).
html/ArvDevice.html:739: warning: no link for: 'FALSE:CAPS' -> (<code class="literal">FALSE</code>).
html/ArvCamera.html:2291: warning: no link for: 'g-object-unref' -> (<code class="function">g_object_unref()</code>).
html/ArvCamera.html:2672: warning: no link for: 'g-free' -> (<code class="function">g_free()</code>).
```

EmmanuelP, 2019-10-17:
Hi,

I guess the documentation of aravis dependencies are not installed (glib mainly). This is mandatory, 
as gtk-doc need them to correctly set the URLs to the documentation of the symbol used by aravis.

lovesegfault, 2019-10-18:
Hi @EmmanuelP,

Indeed this fixed most of the warnings, and I've amended the dependencies in the package.

One warning remains:
```
Unescaped left brace in regex is passed through in regex; marked by <-- HERE in m/(.*?){ <-- HERE / at /usr/bin/gtkdoc-scan line 735.
```

EmmanuelP, 2019-10-18:
Which version of gtkdoc-scan are you using (`gtkdoc-scan --version`) ?

lovesegfault, 2019-10-18:
Aha! Indeed the problem is upstream: 
```
❯ gtkdoc-scan --version
Unescaped left brace in regex is passed through in regex; marked by <-- HERE in m/(.*?){ <-- HERE / at /usr/bin/gtkdoc-scan line 735.
1.25
```

Thanks Emmanuel!

closed 2019-10-18 by lovesegfault