AravisProject/aravis #307: GNU Library GPL vs GNU Lesser GPL

de-vri-es, 2019-10-26:
The `COPYING` contains the GNU **Library** General Public License, but the preamble in source files is for the GNU **Lesser** General Public License.

I didn't even know about the existence of the GNU Library General Public License before now, and it has been deprecated by GNU apparently. I expect it's quite similar to the GNU LGPL. Nevertheless, it's currently not entirely clear which license applies.

Would it be possible to switch the `COPYING` file to the regular `GNU LGPL 2+` (so `Lesser`, not `Library`)? This library is much more actively used and better known.

If it helps, I think the information in the source files is more authoritative anyway, so it probably doesn't even technically count as a license switch. I can also submit a PR to do the switch if preferred.

EmmanuelP, 2019-10-26:
You are right, the COPYING file is wrong. Please submit a PR with the GNU LGPL 2+ text.

Thanks.

de-vri-es, 2019-10-26:
PR opened at #308. Thanks for the quick reply :)

closed 2019-10-26 by de-vri-es

EmmanuelP, 2025-02-14:
Hi @de-vri-es 

FYI, I have reverted the COPYING file to lgpl-2.0-or-later, following an issue report. lgpl-2.0-or-later is compatible with lgpl-2.1-or-later and lgpl-3.0-or-later.

The lesser vs. library wording should not be an issue, as the GNU Lesser General Public License counts as the successor of the GNU Library General Public License. <https://www.gnu.org/licenses/old-licenses/lgpl-2.1.en.html>

de-vri-es, 2025-02-15:
Ok, thanks for letting me know. Shouldn't cause any issues for me :)

EmmanuelP, 2025-02-16:
Well, I have changed again the COPYING file to use LGPL 2.1. The license situation should be much more consistent now. Everything is mentioning lgpl-2.1-or-later (source files, readme, citation file), except the examples that are using unlicensed.

de-vri-es, 2025-02-16:
Haha, ok :)