AravisProject/aravis #310: Unable to compile aravis 0.6 branch on debian testing

TIS-Edgar, 2019-11-12:
Hi Emmanuel,

I am not sure if you still accept bugs for 0.6 releases.
I am unable to compile aravis.

Plattform:
Debian Testing on an amd64

```
git clone https://github.com/AravisProject/aravis.git
mkdir aravis/build
cd aravis/build
git checkout origin/aravis-0-6
 ../autogen.sh --disable-dependency-tracking
```

Calling ```make```results in:

```
Making all in src
make[1]: Entering directory '/home/edt/work/aravis/build_autotools/src'
Makefile:1529: *** missing separator.  Stop.
make[1]: Leaving directory '/home/edt/work/aravis/build_autotools/src'
make: *** [Makefile:594: all-recursive] Error 1
```

The generated Makefile 

```
.PHONY: test test-report perf-report full-report
# run make test as part of make check
check-local: test

@CODE_COVERAGE_RULES@ 
```
Compilation with the same steps succeeds on Ubuntu 18.04 LTS.

If you need any additional information, please let me know.

EmmanuelP, 2019-11-12:
Hi Edgar,

> I am not sure if you still accept bugs for 0.6 releases.

Sure, no problem.

> I am unable to compile aravis.
> Plattform: Debian Testing on an amd64

There is debian package for aravis 0.6.0. In the rules file, there is:

```
	./autogen.sh --enable-gtk-doc --enable-usb --enable-gst-plugin \
	             --enable-viewer --enable-introspection=yes \
	             --enable-packet-socket --enable-static=yes
 ```

What is supposed to do `--disable-dependency-tracking` ?

Could you try to build using the release tarballs from here: https://ftp.gnome.org/pub/gnome/sources/aravis/0.6/ ?

My knowlege of autotools is pretty limited (hence the move to meson in the next aravis series). Anyway, could you attach the entire Makefile ?

TIS-Edgar, 2019-11-13:
> What is supposed to do --disable-dependency-tracking ?

When executing without autogen.sh reports:

```
config.status: executing depfiles commands
config.status: error: in `/home/edt/work/aravis/build_autotools':
config.status: error: Something went wrong bootstrapping makefile fragments
    for automatic dependency tracking.  Try re-running configure with the
    '--disable-dependency-tracking' option to at least be able to build
    the package (albeit without support for automatic dependency tracking).
See `config.log' for more details
```
The complete autogen.sh output and the config.log can be [found here](https://gist.github.com/TIS-Edgar/0fcaf0187b83d1f0f3b65bfdce8b24b6).

The behavior is identical when using the debian package rules. `--disable-dependency-tracking` is required. The generated Makefile is faulty.

You can find the complete Makefile for the initial case [here](https://gist.github.com/TIS-Edgar/106083ce5c333a438cc2fa23e7df0bdf).

I downloaded aravis-0.6.4.tar.xz and executed `./configure` and `make`.
Both executed without error.

EmmanuelP, 2019-11-13:
Well, automake output is not really helpfull. It complains about dependency tracking, while there is no error related to dependency in the log file...

Which version of automake/autoconf are you using ?

EmmanuelP, 2019-11-13:
It looks like you are building outside of the source tree. Could you try to build in tree ?

TIS-Edgar, 2019-11-13:
automake version 1.16.1
autoconf version 2.69
Both are from the debian repositories.

An in-source configuration results in the same behavior. `--disable-dependency-tracking` is required and the generated Makefile is faulty.

EmmanuelP, 2019-11-21:
It looks like autoconf-archive broke their API: https://gitlab.gnome.org/GNOME/libgdata/issues/24

Could you try to remove the following line:

```
aclocal --install -I m4 || exit 1
```

from autogen.sh.

TIS-Edgar, 2019-11-25:
Hi Emmanuel,

Sorry for the late reply, I got struck down with a cold.

Removing the line you mentioned results in a successful configuration and build.
`--disable-dependency-tracking` is also no longer required. I don't know if that is a coincidence or not.

I tried the default configuration, the debian package configuration and my own. They all compile successfully now.

EmmanuelP, 2019-11-25:
`aclocal --install` makes a copy of the system installed m4 files into the aravis m4 directory. But the API of what is on your system differs to the API of the file bundled with aravis.

I have disabled the call to aclocal in autogen.sh in ff48ffe0401fe42a1e8567061af35ceb79b374e2

EmmanuelP, 2019-11-25:
Thanks for the report Edgar.

closed 2019-11-25 by EmmanuelP