AravisProject/aravis #314: Bug during the transfer that prevents the transmission of the complete payload.

mcclauba, 2019-11-21:
**Describe the bug**
Bug during video transfer that prevents the transmission of the complete payload.  Streaming from Allied Vision GeniCam with resolution greater than approx. 350x350 results in "Incomplete image received, dropping"

**To Reproduce**
Using an Allied Vision Alvium 1800-500c camera, attempt the following pipeline:
gst-launch-1.0 -e aravissrc camera-name="Allied Vision-xxxxx" \
! video/x-raw,format=GRAY8,width=2592,height=1944,framerate=4/1 \
! videoconvert \
! ximagesink

**Expected behavior**
A successfully displayed video stream at 4 frames per second

**Camera description:**
 - Manufacturer: Allied Vision
 - Model: Alvium 1800-500c
 - Interface: USB3

**Platform description:**
 - Aravis version: 0.7.3
 - OS: Ubuntu 18.04
 - Hardware:  Nvidia Jetson TX2 dev kit:
        NVIDIA Pascal™ Architecture GPU
        2 Denver 64-bit CPUs + Quad-Core A57 Complex
        8 GB L128 bit DDR4 Memory
        32 GB eMMC 5.1 Flash Storage
   
**Additional context**
When running nearly the same pipeline with far less resolution (i.e. 200x200), the stream works fine.  As the resolution increases, the displayed video becomes more and more choppy as if the USB bandwidth is reaching its maximum capacity.  This threshold I have found to be around 350x350, where greater than this results in the incomplete image errors.  The amount of data in such a stream should not be near enough to bog down USB3.0.  

I have confirmed that my hardware is functioning properly by running a test stream with Allied Vision's example software (Vimba API built on top of GenICam API), which successfully captures frames at full resolution of 2592x1944 and framerates upwards of 24 fps.

[arv-camera-test-output.txt](https://github.com/AravisProject/aravis/files/3874093/arv-camera-test-output.txt)

EmmanuelP, 2019-11-21:
Hi Richard,

Thanks for the report.

Could you rerun your test with this patch applied: 783a2968e540768914877c3487ae5d7719f1d00f

It will display the number of received bytes in case of incomplete transfers. Make sure the debug output is enabled.

mcclauba, 2019-11-21:
After applying your patch...

$ ./arv-camera-test -d all
Looking for the first available camera
Found 1 USB3Vision device (among 3 USB devices)
[UvDevice::new] Vendor  = Allied Vision
[UvDevice::new] Product = ALVIUM 1800 U-500c
[UvDevice::new] S/N     = 001G9
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 5, interface 1
Get genicam
MANUFACTURER_NAME =        'Allied Vision'
MAX_DEVICE_RESPONSE_TIME = 0x0000012c
DEVICE_CAPABILITY        = 0x0000000000000f09
SRBM_ADDRESS =             0x0000000000010000
MANIFEST_TABLE_ADDRESS =   0x0000000000100000
U3VCP_CAPABILITY =         0x00000001
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x0000000000020000
SI_INFO =                  0x0a000000
SI_CONTROL =               0x00000000
SI_REQ_PAYLOAD_SIZE =      0x000000000015f900
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000020
SI_MAX_LEADER_SIZE =       0x00000400
SI_PAYLOAD_SIZE =          0x00100000
SI_PAYLOAD_COUNT =         0x00000001
SI_TRANSFER1_SIZE =        0x0005fc00
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000400
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 00 00 20 00 00 00 00 00  .......... .....
00000010 54 c5 00 00 00 00 00 00 25 81 6f 8c 3b dc a7 08  T.......%.o.;...
00000020 39 9a 0a d9 6c 15 62 6e 9b f7 81 59 00 00 00 00  9...l.bn...Y....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x0000000000200000
genicam size    =          0x000000000000c554
zip file =                 GenICam_Default.xml
........
[ArvCamera:_update_status] [ArvDevice::read_memory] Timeout
vendor name           = Allied Vision
model name            = 1800 U-500c
[ArvCamera:_update_status] node 'DeviceID' not found
device id             = (null)
image width           = 1200
image height          = 1200
horizontal binning    = 0
vertical binning      = 0
payload               = 1440000 bytes
exposure              = 39997.3 µs
gain                  = 7 dB
uv bandwidth limit     = 200000000 [32750000..450000000]
SI_INFO            =       0x0a000000
SI_REQ_PAYLOAD_SIZE =      0x000000000015f900
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000020
Required alignment =       1024
SI_PAYLOAD_SIZE =          0x00100000
SI_PAYLOAD_COUNT =         0x00000001
SI_TRANSFER1_SIZE =        0x0005fc00
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_LEADER_SIZE =       0x00000400
SI_MAX_TRAILER_SIZE =      0x00000400
USB transfer error: LIBUSB_ERROR_IO
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277960 / expected 1440000)
Incomplete image received, dropping (received 277944 / expected 1440000)
Incomplete image received, dropping (received 277944 / expected 1440000)
Incomplete image received, dropping (received 277952 / expected 1440000)
Incomplete image received, dropping (received 277952 / expected 1440000)
Incomplete image received, dropping (received 277944 / expected 1440000)
Incomplete image received, dropping (received 277960 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277940 / expected 1440000)
Incomplete image received, dropping (received 277940 / expected 1440000)
Incomplete image received, dropping (received 277960 / expected 1440000)
Incomplete image received, dropping (received 277944 / expected 1440000)
Incomplete image received, dropping (received 277960 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277956 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277952 / expected 1440000)
^CIncomplete image received, dropping (received 277956 / expected 1440000)
Incomplete image received, dropping (received 277956 / expected 1440000)
Incomplete image received, dropping (received 277960 / expected 1440000)
Incomplete image received, dropping (received 277952 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Incomplete image received, dropping (received 277948 / expected 1440000)
Frame rate = 0 Hz
Completed buffers = 0
Failures          = 26
Underruns         = 0
Incomplete image received, dropping (received 277960 / expected 1440000)
[UvStream::finalize] n_completed_buffers    = 0
[UvStream::finalize] n_failures             = 27
[UvStream::finalize] n_underruns            = 0
[Stream::finalize] Flush 49 buffer[s] in input queue
[Stream::finalize] Flush 1 buffer[s] in output queue

EmmanuelP, 2019-11-21:
The bug is due to the naive USB transfer parameters computation done in [arvuvstream.c](https://github.com/AravisProject/aravis/blob/783a2968e540768914877c3487ae5d7719f1d00f/src/arvuvstream.c#L252-L337)

To be compared to what is done in the National Instruments driver: https://github.com/ni/usb3vision/blob/ac1599d4ef58c61863839504adff37eaf61810a3/u3v_stream.c#L279

EmmanuelP, 2019-11-21:
Could you try to change the value of ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE and set it to 256*256 ?

https://github.com/AravisProject/aravis/blob/783a2968e540768914877c3487ae5d7719f1d00f/src/arvuvstream.c#L38

mcclauba, 2019-11-21:
Making said change enables a stream of 1200x1200 @ 25fps with arv-camera-test.  It does not change the result of the gst pipeline however.

arv-camera-test output:
vendor name           = Allied Vision
model name            = 1800 U-500c
[ArvCamera:_update_status] node 'DeviceID' not found
device id             = (null)
image width           = 1200
image height          = 1200
horizontal binning    = 0
vertical binning      = 0
payload               = 1440000 bytes
exposure              = 39997.3 µs
gain                  = 7 dB
uv bandwidth limit     = 200000000 [32750000..450000000]
SI_INFO            =       0x0a000000
SI_REQ_PAYLOAD_SIZE =      0x000000000015f900
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000020
Required alignment =       1024
SI_PAYLOAD_SIZE =          0x00010000
SI_PAYLOAD_COUNT =         0x00000015
SI_TRANSFER1_SIZE =        0x0000fc00
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_LEADER_SIZE =       0x00000400
SI_MAX_TRAILER_SIZE =      0x00000400
USB transfer error: LIBUSB_ERROR_IO
Frame rate = 18 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 24 Hz
Frame rate = 24 Hz
Frame rate = 25 Hz
Incomplete image received, dropping (received 1260980 / expected 1440000)
Frame rate = 23 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
Frame rate = 25 Hz
^CFrame rate = 25 Hz
Completed buffers = 590
Failures          = 1
Underruns         = 0
USB transfer error: LIBUSB_ERROR_TIMEOUT
[UvStream::finalize] n_completed_buffers    = 592
[UvStream::finalize] n_failures             = 1
[UvStream::finalize] n_underruns            = 0
[Stream::finalize] Flush 50 buffer[s] in input queue
[Stream::finalize] Flush 0 buffer[s] in output queue

EmmanuelP, 2019-11-21:
Ok, that is better, even if the incomplete warning is a bit worrying.

We have to find how to determine the optimum maximum transfer value, instead of using a `#define`.

mcclauba, 2019-11-21:
What is the UV stream by the way?

EmmanuelP, 2019-11-21:
I'm not sure to understand the question. ArvUvStream derives from ArvStream, and implements the USB3Vision streaming protocol.

mcclauba, 2019-11-24:
A better question would have been how does ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE affect the stream.  As the name suggests, it defines the USB3vision maximum transfer size, simple enough.  What puzzled me was how reducing this maximum from 2^20 to 2^16 made for successfully completed buffers when running arv-camera-test.   I can test a range of values tomorrow, but it will be just blind guessing.

With my camera, the uv bandwidth limit can be controlled, but the current value of 200 MB is greater than the payload in the test stream of 1.44 MB.  So I assume that the bandwidth limit is something else than the stream's max transfer size.

EmmanuelP, 2019-11-25:
> A better question would have been how does ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE affect the stream.

This value is used for the computation of the USB3Vision streaming parameters.

https://github.com/AravisProject/aravis/blob/783a2968e540768914877c3487ae5d7719f1d00f/src/arvuvstream.c#L294-L314

It looks like on some system, there is a maximum transfer size that should not be exceeded. But I don't know how to determine this limit.

mcclauba, 2019-11-28:
I've forwarded this issue to Allied Vision.  This is their response:

_DeviceLinkThroughputLimit in Byte/s effectively controls the transfer bandwidth.
It needs to be decreased down below 200 or 100 MB/s depending on the adapter and system performance to avoid missing packets and/or dropped frames.
Please let us know if it's helpful on this matter._

I will perform some testing with various levels of device link throughput in attempts to achieve a successful stream.  I'll keep you posted.

mcclauba, 2019-12-03:
Hi again,

I've done some testing which found that by decreasing both the _#define ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE_ inarvuvstream.c to 256x256 and setting my camera's DeviceLinkThroughputMode to 'On' and the DeviceLinkThroughputLimit to less than 185 000 000 bytes allowed for successfull streaming at 2592x1944, RGB, @ 4 fps.  

Increasing the link throughput limit to allow for 25 fps is too much for my USB port, and thus causes missed frames.  This happens with Allied Vision's simple frame grabber C program, so is not the fault of Aravis.

I have found that if _#define ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE_ is set to its original value of 1048576 while setting the DeviceLinkThroughputLimit still amounts to missed frames.  

Regarding this, I believe that having a aravissrc property that sets the device link throughput limit would be beneficial.   Also, as a side note, I know that my camera when in ExposureAutoMode=Continuous it has a tendency to increase the exposure so much that the frame rate decreases.  Could a plugin property be added which allows for the control of exposure auto limits?

EmmanuelP, 2019-12-03:
Thanks for your feedback.

Regarding aravissrc, I think we need a new property that would take a string containing a list a feature/value pairs, similar to the parameters of the arv-tool `control` command. I have created a new issue about this feature: https://github.com/AravisProject/aravis/issues/319

PolyVinalDistillate, 2024-03-13:
Hello - I am involved with the University of Aberdeen on a project where we use various monochrome machine vision cameras for the purposes of recording digital holograms of plankton. I used Aravis orginally under Linux to implement a holographic camera (weeHoloCam) when it was in release 0.6. with no issues, however I'm now using it under Windows for holography with an collection of different sensors.

I have found that various high-spec windows laptops have a tendency to drop frames with "ARV_BUFFER_STATUS_MISSING_PACKETS" and similar issues while other laptops are fine. This seems to be at least partly computer dependant. I had assumed it was bottlenecks in the USB subsystems, but now I'm not at all sure.

Due to the apparent dependency on different computers as well as different cameras, I would like to advocate the exposure of the ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE parameter via your API so it can be modified at runtime. Given the issues I've been experiencing, I would be inclined to implement an auto-tune of this parameter if I could access it at runtime.

For example, we have one laptop that works fine using a Jai GO-5100M with ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE set to 1024*1024, but on a higher spec'd laptop I have had to reduce this to 256*256 to get the sensor working reliably. As the syssem is currently with a customer, I have had to recompile aravis and have them replace libaravis-0.8-0.dll progressively until they found a version that worked.

DxTa, 2024-11-01:
fyi, to make it work with basler cameras, I do need to change `ARV_UV_STREAM_MAXIMUM_SUBMIT_TOTAL` as well.

closed 2025-02-16 by EmmanuelP