AravisProject/aravis #315: gc_boolean_get_value boolean return value conflicts with glib error handling guidelines

de-vri-es, 2019-11-21:
The function `gc_boolean_get_value` take a `GError **` and it returns a boolean, but the return value doesn't indicate if the function call failed. This is in contradiction with the glib error handling guidelines [1], which says you should use an output parameter for the boolean instead:
> By convention, if you return a boolean value indicating success then TRUE means success and FALSE means failure. **Avoid creating functions which have a boolean return value and a GError parameter, but where the boolean does something other than signal whether the GError is set**. Among other problems, it requires C callers to allocate a temporary error. Instead, provide a "gboolean *" out parameter. There are functions in GLib itself such as g_key_file_has_key() that are deprecated because of this.

The result is that the automatically generated bindings assume there is no data in the return value. In Rust, that translates to a `Result` with no value.

Problem is of course that changing the return value to an output parameter is API incompatible. And it would break symmetry with the other nodes that simply return their value.

I'd be willing to submit a PR if we can decide on a solution. Also if that means changing all `gc_*_get_value` functions to use output parameters.

[1] https://developer.gnome.org/glib/stable/glib-Error-Reporting.html

EmmanuelP, 2019-11-21:
Hi Maarten,

Glib has a function similar to arv_gc_boolean_get_value, g_key_file_get_boolean. 

I'm not familiar with glib rust bindings, but it looks like it is handled as a special case and bound manually: http://gtk-rs.org/docs/src/glib/key_file.rs.html#98

EmmanuelP, 2019-11-21:
IMHO, it is a [GIR](https://github.com/gtk-rs/gir) bug. It should handle the case where there is no output value, except the returned boolean. May be with a special annotation.

Anyway, I don't want to break the consistency with the rest of the C API. And I also don't want to change the rest of the API to make it use an output parameter, that makes the C API ugly and inconvenient.

A possible solution, but I'm not sure if it works, would be to declare another function with a boolean output parameter, tell gobject introspection to rename it to gc_boolean_get_value and to ignore the real gc_boolean_get_value.

EmmanuelP, 2019-11-21:
The `rename-to` annotation should do the trick. There is even no need to tell gobject introspection to ignore the original function.

https://wiki.gnome.org/Projects/GObjectIntrospection/Annotations

de-vri-es, 2019-11-21:
> Glib has a function similar to arv_gc_boolean_get_value, g_key_file_get_boolean.

True, although that function has been deprecated because it doesn't follow the error handling guidelines.

> IMHO, it is a [GIR](https://github.com/gtk-rs/gir) bug. It should handle the case where there is no output value, except the returned boolean. May be with a special annotation.

Problem is that the gir tool can not distinguish it from a function that doesn't have any return value, but only performs an action (and returns a boolean to indicate failure). As far as I could tell, there is no annotation to indicate the difference.

> The `rename-to` annotation should do the trick. There is even no need to tell gobject introspection to ignore the original function.

Interesting. Would you be open to a PR that adds such a function? If so, I can try it locally first and make sure it works as intended.

I do have a manual wrapper as workaround now. But because of the way the gtk-rs gir tool works, this is slightly annoying to use (it's in a separate trait that needs to be brought into scope).

Another option is to see if gtk-rs will accept a PR to support functions with boolean return values that don't signal an error. Since there doesn't seem to be an annotation for it, it would have to be something specific for the gtk-rs tool.

If you prefer I try fixing this in the gtk-rs gir tool, I can understand that too.  In the mean time, I have the manual wrapper as workaround.

de-vri-es, 2019-11-21:
Hmm, it appears the functions are not deprecated, even though the error handling guideline explicitly mentions `g_key_file_has_key` is deprecated for that reason. The source code seems to disagree. `g_key_file_has_key` only has a comment, no deprecation annotation, and there doesn't seem to be an alternative to `g_key_file_get_boolean`.

EmmanuelP, 2019-11-24:
@de-vri-es Please test c6e238e67260e5925b488e3fa76492d91c5ff0f1 and check if that fixes the automatic rust binding.

de-vri-es, 2019-11-25:
It works, with some additional configuration of `gtk-rs/gir`. The tool doesn't automatically honor the `rename-to` annotation, but it can be configured to do the rename (since about a week). So that's good enough for me :)

EmmanuelP, 2019-11-25:
Ok. Pushed to master. Thanks.

closed 2019-11-25 by EmmanuelP

de-vri-es, 2019-11-25:
Thanks for the patch :)

EmmanuelP, 2019-11-27:
Maarten, don't hesitate to open more issues when you think the C API can be improved for better bindings.

Thanks again.

de-vri-es, 2019-11-27:
Will do. And thanks again for the project. It's so much nicer than using the different proprietary genicam wrappers from each manufacturer.

commit 06794a43 (2019-11-25): introspection: workaround binding issue of boolean returning functions (#316)

The function gc_boolean_get_value take a GError ** and it returns a
boolean, but the return value doesn't indicate if the function call
failed. The result is that the automatically generated bindings assume
there is no data in the return value. In Rust, that translates to a
Result with no value.

Fix the binding by implementing functions taking an ouput parameter, and
using rename-to annotation to replace the functions designed for C use.

Fixes #315