AravisProject/aravis #318: aravis support timeframes

TIS-Edgar, 2019-12-02:
Hi Emmanuel,

Thank you for your input concerning future migrations to aravis 0.7.
Concerning this I have some questions to find the best way to ensure long term integration of aravis in tiscamera.

We have to support Ubuntu 18.04 LTS until at least 2021.
With environments like the NVidia platforms (TX2, etc) this is not really negotiable for us. 

Aravis master is at this point in time able to compile on Ubuntu 18.04 LTS
when the meson version is reduced to 0.45 and tests are disabled (due to `pymod = import ('python')`) which requires 0.47.
I expect the changes that prevent aravis from compiling
on our reference platform to increase as time goes on.
From our point it would be desirable to always be as near as possible to a vanilla version of either aravis master or the latest aravis stable release especially considering USB3Vision support.

How long will support for the aravis-0.6 branch exist?
Are there compatibility promises or reference systems one can adhere to?

Cheers and thank you for your time and effort,
Edgar

EmmanuelP, 2019-12-02:
Hi Edgar,
Le lun., déc. 2, 2019 at 05:08, Edgar <notifications@github.com> a 
écrit :
> Concerning this I have some questions to find the best way to ensure
> long term integration of aravis in tiscamera.
>
> We have to support Ubuntu 18.04 LTS until at least 2021.
>  With environments like [...]
The 0.7 branch has already diverged a lot of the 0.6 branch, so I'm not 
going to systematically backport the 0.7 bugfixes to 0.6. But if I have 
issues opened for the 0.6.x branch, or even better, pull requests, I 
will take time to look at them, fix the branch, and make somes releases.

> Are there compatibility promises or reference systems one can adhere 
> to?
> 
Well, not really. But I try to not bump dependency versions without a 
good reason. I try to use versions available in the latest LTS releases 
of the major distribution (ubuntu, Redhat, CentOS). For aravis itself 
(the library), Glib requirement is 2.44, which is almost 5 year old. 
The viewer requires Gtk+ 3.12, wich is also 5 year old. I'm not 
planning to change the requirements for aravis 0.7/0.8.

Also, the set of dependencies is quite limited, so even on older 
system, it is possible to install aravis and its dependencies in a 
custom prefix.

Regarding meson, I guess I can try to lower the version requirement to 
0.45, I don't think aravis relies too much on newer features.

Cheers,

    Emmanuel.

EmmanuelP, 2019-12-02:
I forgot to tell you that for this sort of questions, I prefer to have them on the aravis discourse instance, instead of in a github issue.

Cheers,

  Emmanuel.

closed 2019-12-02 by EmmanuelP