AravisProject/aravis #319: Add a genicam feature control property to aravis gstreamer plugin

EmmanuelP, 2019-12-03:
Sometimes, using the gstreamer plugin, it is necessary to set some device feature in order to make it work properly.

A possible implementation would be to add a `features` property, that takes a list of feature/value pairs, similar to the parameters of the `control` command of arv-tool.

Then we could write:

```
gst-launch-1.0 aravissrc features="DeviceLinkThroughputLimit=185000000 ShutterMode=Rolling" ! videoconvert ! autovideosink
```

mcclauba, 2019-12-04:
A very good idea.  Just need to be sure that a few features are required to be in the correct mode for their values to be manually set.

barzan-hayati, 2020-02-15:
> Sometimes, using the gstreamer plugin, it is necessary to set some device feature in order to make it work properly.
>
> A possible implementation would be to add a `features` property, that takes a [...]

Hi @EmmanuelP . `Features` command does not work for me.
`gst-launch-1.0 aravissrc features="DeviceLinkThroughputLimit=185000000 ShutterMode=Rolling" ! videoconvert ! autovideosink`

`WARNING: erroneous pipeline: no property "features" in element "aravis0"
`

and by adding aravis source address:

` gst-launch-1.0 -e --gst-plugin-path=/usr/local/lib/ aravissrc features="DeviceLinkThroughputLimit=185000000 ShutterMode=Rolling" ! videoconvert ! autovideosink`

`WARNING: erroneous pipeline: no property "features" in element "aravis0"
`

EmmanuelP, 2020-04-06:
@barzan-hayati it is not implemented yet...

EmmanuelP, 2020-04-06:
Implemented in a755517e44b1190379e151c7403dc861ce88bedf

closed 2020-04-06 by EmmanuelP

nrstrifork, 2022-07-26:
Hi @EmmanuelP,

Did you document all genicam features, or are you using the same feature key/val naming as genicam?

EmmanuelP, 2022-07-26:
Hi,

> Did you document all genicam features, or are you using the same feature key/val naming as genicam?

I fail to understand what you mean. Could you rephrase your question ?

EmmanuelP, 2022-07-26:
You can set any feature available on your camera, using the same syntax as the arv-tool utility. You can list every available features using `arv-tool-0.8 features`.

nrstrifork, 2022-07-26:
> You can set any feature available on your camera, using the same syntax as the `control` command of `arv-tool`. You can list every available features using `arv-tool-0.8 features`.

Thank you :)