AravisProject/aravis #321: Get in Error by make viewer

barzan-hayati, 2019-12-08:
**Describe the bug**
Couldn't run arv-viewer  to watch live video from USB Camera.

**To Reproduce**
I installed [aravis 0.6.4](http://ftp.acc.umu.se/pub/GNOME/sources/aravis/0.6/aravis-0.6.4.tar.xz) using these commands. 
```
./configure
make
make install
```
It was make successfully, and I could test aravis via other commands such as :
```
tests/arv-camera-test -f 30 - e 30000 -d all
tests/arv-camera-test --debug all:3
src/arv-tool-0.6
tests/arv-camera-test -d all
arv-tool-0.6 features
./arv-camera-test
arv-tool-0.6 features -d all:3
sudo ./arv-camera-test -d all:3
sudo arv-tool-0.6 -d all:3
sudo arv-tool-0.6 -d all:3 features
arv-camera-test -d all:3
arv-tool-0.6 features
arv-tool-0.6 --debug=all:3
sudo lsusb -v
tests/arv-device-test
src/arv-tool-0.6 --debug=all
tests/arv-device-test
tests/arv-camera-test
arv-tool-0.6 genicam

```
**Expected behavior**
Now, I want to use viewer tools to watching live video from your tools by running this command:
`./arv-viewer -d all:3 `
But there is not any arv-viewer files in aravis directory. I know that there is a Makefile in **viewer** and **gst-0.10** directories and I tried to make them. But, by run **make** command in them , I received  these messages as error and it could not make and create **arv-viewer**
```
Making all in data
make[1]: Entering directory '/home/nano/Downloads/Aravis/aravis-0.6.4/viewer/data'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/nano/Downloads/Aravis/aravis-0.6.4/viewer/data'
make[1]: Entering directory '/home/nano/Downloads/Aravis/aravis-0.6.4/viewer'
  CC       main.o
In file included from ./arvviewer.h:1:0,
                 from main.c:24:
./arvviewertypes.h:26:10: fatal error: glib-object.h: No such file or directory
 #include <glib-object.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
Makefile:627: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 1
make[1]: Leaving directory '/home/nano/Downloads/Aravis/aravis-0.6.4/viewer'
Makefile:786: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1

```

**Camera description:**
 - Manufacturer: Hikvision (U3V)
 - Model: MV-CA050-20UC
 - Interface: USB3

**Platform description:**
 - Aravis version: 0.6.4
 - OS: ubuntu bionic 18.04
 - Hardware: Jetson Nano(arm64)

**Additional context**
As mentioned in error message, I could not access  `glib-object.h` but I added it to list of variable path in ubuntu. 
**`echo $PATH`**
`/usr/lib/glib-2.0/include:/usr/include/gstreamer-1.0/gst:/usr/include/gstreamer-1.0/gst:/usr/lib/glib-2.0/include:/usr/include/gstreamer-1.0/gst:/usr/include/gstreamer-1.0/gst:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/include/gstreamer-1.0/gst:/usr/include/glib-2.0:/usr/lib/glib-2.0/include:/usr/lib/aarch64-linux-gnu/glib-2.0/include:/usr/include/gstreamer-1.0/gst:/usr/include/gstreamer-1.0:/usr/lib/aarch64-linux-gnu/glib-2.0/include:/usr/include/gstreamer-1.0/gst:/usr/include/glib-2.0:/usr/lib/glib-2.0/include:/usr/lib/aarch64-linux-gnu/glib-2.0/include:/usr/include/gstreamer-1.0/gst:/usr/include/gstreamer-1.0:/usr/lib/aarch64-linux-gnu/glib-2.0/include
`

Also for **gst-0.10** directory, I want to make but this error happened:
```
  CC       libgstaravis_0.6_la-gstaravis.lo
In file included from gstaravis.c:37:0:
./gstaravis.h:26:10: fatal error: gst/gst.h: No such file or directory
 #include <gst/gst.h>
          ^~~~~~~~~~~
compilation terminated.
Makefile:574: recipe for target 'libgstaravis_0.6_la-gstaravis.lo' failed
make: *** [libgstaravis_0.6_la-gstaravis.lo] Error 1

```

While I'm sure that  `gst/gst.h` is in the list of my path.
Thanks in advance.

EmmanuelP, 2019-12-08:
> ./configure
make
make install

What is the output of `./configure` ?

The viewer is probably disabed. Try `configure --enable-viewer --enable-gst-plugin`.

barzan-hayati, 2019-12-09:
Output of **configuration**
> ./configure

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for library containing strerror... none required
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking locale.h usability... yes
checking locale.h presence... yes
checking for locale.h... yes
checking for LC_MESSAGES... yes
checking for CFPreferencesCopyAppValue... no
checking for CFLocaleCopyCurrent... no
checking libintl.h usability... yes
checking libintl.h presence... yes
checking for libintl.h... yes
checking for ngettext in libc... yes
checking for dgettext in libc... yes
checking for bind_textdomain_codeset... yes
checking for msgfmt... /usr/bin/msgfmt
checking for dcgettext... yes
checking if msgfmt accepts -c... yes
checking for gmsgfmt... /usr/bin/msgfmt
checking for xgettext... /usr/bin/xgettext
checking whether NLS is requested... yes
checking for intltool >= 0.31.2... 0.51.0 found
checking for intltool-update... /usr/bin/intltool-update
checking for intltool-merge... /usr/bin/intltool-merge
[... 103 lines cut ...]
config.status: creating docs/Makefile
config.status: creating docs/reference/Makefile
config.status: creating docs/reference/aravis/Makefile
config.status: creating aravis.pc
config.status: creating src/arvconfig.h
config.status: executing depfiles commands
config.status: executing default-1 commands
config.status: executing libtool commands
config.status: executing po/stamp-it commands

Configuration:

  Compiler:                     gcc
  Compiler flags:               -pthread -I/usr/include/libxml2 -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/libusb-1.0
  Linker flags:                 -lm -lgio-2.0 -lgobject-2.0 -lxml2 -lgthread-2.0 -pthread -lglib-2.0 -lz -lusb-1.0

  Build USB support:            yes
  Build packet socket support:  no
  Build viewer:                 no
  Build gstreamer plugin:       yes
  Build gstreamer-0.10 plugin:  no
  Build with fast heartbeat:    no
  Build with address sanitizer: no

```

barzan-hayati, 2019-12-09:
Output of first try **`./configure --enable-viewer --enable-gst-plugin`**

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... no
checking for mawk... mawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking whether make supports nested variables... (cached) yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for library containing strerror... none required
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking locale.h usability... yes
checking locale.h presence... yes
checking for locale.h... yes
checking for LC_MESSAGES... yes
checking for CFPreferencesCopyAppValue... no
checking for CFLocaleCopyCurrent... no
checking libintl.h usability... yes
checking libintl.h presence... yes
checking for libintl.h... yes
checking for ngettext in libc... yes
checking for dgettext in libc... yes
checking for bind_textdomain_codeset... yes
checking for msgfmt... /usr/bin/msgfmt
checking for dcgettext... yes
checking if msgfmt accepts -c... yes
checking for gmsgfmt... /usr/bin/msgfmt
checking for xgettext... /usr/bin/xgettext
checking whether NLS is requested... yes
checking for intltool >= 0.31.2... 0.51.0 found
checking for intltool-update... /usr/bin/intltool-update
checking for intltool-merge... /usr/bin/intltool-merge
checking for intltool-extract... /usr/bin/intltool-extract
[... 394 lines cut ...]
config.status: creating docs/Makefile
config.status: creating docs/reference/Makefile
config.status: creating docs/reference/aravis/Makefile
config.status: creating aravis.pc
config.status: creating src/arvconfig.h
config.status: executing depfiles commands
config.status: executing default-1 commands
config.status: executing libtool commands
config.status: executing po/stamp-it commands

Configuration:

  Compiler:                     gcc
  Compiler flags:               -pthread -I/usr/include/libxml2 -I/usr/include/glib-2.0 -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/libusb-1.0
  Linker flags:                 -lm -lgio-2.0 -lgobject-2.0 -lxml2 -lgthread-2.0 -pthread -lglib-2.0 -lz -lusb-1.0

  Build USB support:            yes
  Build packet socket support:  no
  Build viewer:                 no
  Build gstreamer plugin:       yes
  Build gstreamer-0.10 plugin:  no
  Build with fast heartbeat:    no
  Build with address sanitizer: no

```

EmmanuelP, 2019-12-09:
> configure: error: No package 'gtk+-3.0' found
No package 'libnotify' found - Can not build viewer

If you want to build the viewer, you must install the development packages for gtk+3 and libnotify.

Cheers,

Emmanuel.

closed 2019-12-09 by EmmanuelP

barzan-hayati, 2019-12-09:
As a point: I have installed **gstreamer-1.0** before installing aravis(shown in variables path).

EmmanuelP, 2019-12-09:
> As a point: I have installed gstreamer-1.0 before installing aravis(shown in variables path).

Yes, and that is why you can build the gstreamer plugin.

barzan-hayati, 2019-12-09:
> > configure: error: No package 'gtk+-3.0' found
> > No package 'libnotify' found - Can not build viewer
> 
> If you want to build the viewer, you must install the development packages for gtk+3 and libnotify.
> 
> Cheers,
> 
> Emmanuel.

Thanks. I installed(build from source) gtk and arv-viewer worked for me.