AravisProject/aravis #324: Bad USB3Vision performance on Nvidia Xavier board

EmmanuelP, 2019-12-17:
As reported in #323 , USB3Vision maximum framerate is way lower than what is possible using Baslers's Pylon software on a NVidia Xavier board.

@mikekml , what is the output of lsmod ? Does Pylon USB3Vision support use a dedicated kernel module ?

mikekml, 2019-12-18:
@EmmanuelP , see below lsmod output. As far as i can see Pylon is not using proprietary kernel driver to access the camera.  Pylon is using libusb to access the camera - confirmed with setting LIBUSB_DEBUG=4 .. could clearly trace that Pylon is using async bulk request handling. 

lsmod 
Module                  Size  Used by
usbmon                 26883  6
nfnetlink_queue        14050  0
nfnetlink_log          11033  0
nfnetlink               7959  2 nfnetlink_log,nfnetlink_queue
bnep                   16562  2
fuse                  103841  7
overlay                48691  0
zram                   26166  8
nvgpu                1569917  94
bluedroid_pm           13912  0
ip_tables              19441  0
x_tables               28951  1 ip_tables

EmmanuelP, 2019-12-19:
https://github.com/AravisProject/aravis/issues/323#issuecomment-566761645

EmmanuelP, 2019-12-19:
Hi @mikekml @crabsmaps 

I've spent some time yesterday in testing the old async usb [branch](https://github.com/AravisProject/aravis/tree/wip/async-usb) 

On a Jetson TX2 board, it really makes a big difference, allowing a steady 380MB/s transfer. This is achieved by not only using libusb async API, but also because the patch piles USB bulk transfer requests.

On a standard laptop (Intel Core i5-5300U CPU @ 2.30GHz × 4 ), while the CPU use seems to increase a bit for comparable bandwiths, the patch allows to obtain the best maximum transfer speed.

So it is worth porting the patch to master.

makelinux, 2019-12-24:
Hi @EmmanuelP, I've rebased branch usb-async on master version 7. It's compiled but doesn't work. I haven't resources to debug it.  I hope it can help you: https://github.com/makelinux/aravis/tree/async-usb-7
Regards

crabsmaps, 2020-01-02:
We don't have any high-fps usb cameras at hand, but we tested async usb branch with two cameras (basler aca4096-30uc and basler acA1920-40uc) on one of the Xilinx boards. Behavior was essentially the same as with 0.6.4 branch.

makelinux, 2020-01-07:
To see improvements of async need to use slower computer or run stress test.

makkarpov, 2020-12-07:
I confirm that Aravis thru USB is unusable on Jetson Xavier board.

We have [this](https://www.alliedvision.com/en/products/embedded-vision-cameras/detail/Alvium%201800%20U/-240/closedhousing-c-mount-standard-color.html) camera, which is neither high-resolution (2.3 MP) nor high-FPS (80 is the maximum).

Aravis works fine with this camera on desktop PC, however it cannot acquire any image on Jetson board. It can work at extremely low resolutions (like 300x300) or extremely low FPS's, but never at full resolution and reasonable FPS. 

Proprietary vendor software works fine at full resolution and full FPS.

makelinux, 2020-12-07:
You can find a lot of custom improvements on my fork: https://github.com/makelinux/aravis

makkarpov, 2020-12-08:
Build instructions would be nice, especially about very weird `ctracer.h` and `lookup_symbol_name`.

I found your `ldt` project with ctracer.h file, but I cannot figure out how linux driver is related to the userspace aravis lib and where can I get corresponding symbol.

EmmanuelP, 2021-12-17:
We can close this issue now the async-usb work is merged.

closed 2021-12-17 by EmmanuelP