AravisProject/aravis #325: Unknown status (0x4004) on trailer packets

jajberni, 2019-12-19:
**Describe the bug**
I am having troubles getting steady framerates from the camera, regardless of the resolution and framerate. 

The debug output shows that I am getting trailer packets with an unknown status code (0x4004), which discard the frame:

```[Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::new] Legacy endianess handling = yes
[ArvCamera:_update_status] node 'ChunkModeActive' not found (Status changed)
vendor name           = Photonfocus AG
model name            = MV1-D2048x1088-HS05-96-G2-10
device id             = ...
image width           = 2048
image height          = 1088
horizontal binning    = 1
vertical binning      = 1
payload               = 2228224 bytes
exposure              = 2511.56 µs
gain                  = 1 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 1476 bytes
[GvDevice::create_stream] Number of stream channels = 1
[GvStream::stream_new] Packet size = 1476 byte(s)
[GvStream::stream_new] Destination stream port = 60153
[GvStream::stream_new] Source stream port = 20202
[GvStream::stream_thread] Packet timeout = 20 ms
[GvStream::stream_thread] Frame retention = 100 ms
[GvStream::loop] Standard socket method
[GvStream::process_packet] Error packet at dt = 50162, packet id = 1549 frame id = 4
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::process_packet] Error packet at dt = 48492, packet id = 1549 frame id = 5
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::process_packet] Error packet at dt = 47343, packet id = 1549 frame id = 6
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Timeout for frame 4 at dt = 150164
[GvStream::process_packet] Error packet at dt = 47604, packet id = 1549 frame id = 7
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Timeout for frame 5 at dt = 148704
[GvStream::process_packet] Error packet at dt = 47733, packet id = 1549 frame id = 8
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Timeout for frame 6 at dt = 147569
[GvStream::process_packet] Error packet at dt = 49136, packet id = 1549 frame id = 9
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Timeout for frame 7 at dt = 147979
[GvStream::process_packet] Error packet at dt = 48062, packet id = 1549 frame id = 10
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Timeout for frame 8 at dt = 147774
[GvStream::process_packet] Error packet at dt = 48208, packet id = 1549 frame id = 11
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)

[GvStream::check_frame_completion] Timeout for frame 9 at dt = 149481
[GvStream::process_packet] Error packet at dt = 47749, packet id = 1549 frame id = 12
packet_type  =   (null) (0x4004)
content_type =  trailer (0x0002)
...
[GvStream::finalize] n_completed_buffers    = 3
[GvStream::finalize] n_failures             = 34
[GvStream::finalize] n_timeouts             = 34
[GvStream::finalize] n_aborteds             = 3
[GvStream::finalize] n_underruns            = 0
[GvStream::finalize] n_missing_frames       = 0
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 61690
[GvStream::finalize] n_missing_packets      = 34
[GvStream::finalize] n_error_packets        = 36
[GvStream::finalize] n_ignored_packets      = 0
[GvStream::finalize] n_resend_requests      = 155
[GvStream::finalize] n_resent_packets       = 151
[GvStream::finalize] n_duplicated_packets   = 0
[Stream::finalize] Flush 47 buffer[s] in input queue
[Stream::finalize] Flush 3 buffer[s] in output queue
```

I have confirmed this using Wireshark on MacOS and monitoring the packets, but couldn't find any reason. Also, I haven't found any information about that STATUS code. 

**To Reproduce**
Simply start streaming images. The occurrence of this status in the trailer messages seems quite random. 

**Camera description:**
 - Manufacturer: Photonfocus
 - Model: MV1-D2048x1088-HS05-96-G2-10
 - Interface: GigE 

**Platform description:**
 - Aravis version: 0.8 ( aravis-0.7.3 build )
 - OS: Jetpack [L4T 32.2.3] / Ubuntu 18.04 provided by Nvidia JetPack
 - Hardware: Jetson TX2

**Additional context**
The Photonfocus camera is based on Pleora eBUS, but haven't found any documentation of this status code.

EmmanuelP, 2019-12-19:
HI @jajberni ,

Could you attach the relevant part of the wireshark capture (the packet with the strange status code, and a few packets before ad after) ?

Thanks.

jajberni, 2019-12-19:
Hi @EmmanuelP,

Here are two packets, one is weird and the other one is right:
[image]

[image]

Here is a subset of my capture:

[photonfocus_capture_small.pcapng.gz](https://github.com/AravisProject/aravis/files/3985235/photonfocus_capture_small.pcapng.gz)

EmmanuelP, 2019-12-20:
It seems there is nothing we can do on aravis side. I don't really want to add a random status code to the list of valid ones.

Could you contact PhotonFocus and ask them about this issue ?

EmmanuelP, 2020-04-06:
@jajberni any news regarding this issue ?

EmmanuelP, 2020-08-13:
Looking at the wireshark dissector, it looks like the status code is an error code only if bit 31 is set. May be we can safely ignore status code when this bit is not set.

commit 91e9932c (2020-08-13): gvstream: accept more packet types

Packet type indicates an error only if bit 15 is set.

Fixes #325
closed 2020-08-13 by EmmanuelP