AravisProject/aravis #327: (arv-viewer: ): Aravis-WARNING **: Failed to make stream thread high priority

EasitMickly, 2019-12-29:
Hi, i build the source files and it success, but i run the program and it shows (arv-viewer): Aravis-WARNING **: Failed to make stream thread high priority. I don't know how to slove this problem, please help me. Thank you.

**Camera description:**
 - IMI-Tech
 - IMB-711G
 - GIGE

**Platform description:**
 - 0.6 and 0.7
 - Ubuntu 18.11 LTS ARM64
 - Jeson nano

[image]

AntonioMichaelAlvarez, 2020-01-15:
Try to add `cap_sys_nice` permission to arv-viewer executable:
```
sudo setcap cap_sys_nice+ep $(which arv-viewer)
```
This will allow rtkit (or whatever Aravis uses) to increase thread priority.

EmmanuelP, 2020-04-06:
@EasitMickly , did @AntonioMichaelAlvarez advice help you fix your issue ?

EmmanuelP, 2020-08-19:
I'm closing this issue. Please reopen if you have additional informations.

Cheers.

closed 2020-08-19 by EmmanuelP