AravisProject/aravis #330: Low Frame Rate Image Grabbing

barzan-hayati, 2020-01-07:
**Describe the bug**
I want to capture Image using [pypylon Basler](https://github.com/basler/pypylon) tools. This tools capture images using python. But it take 0.11 sec for grabbing each image for a 3 M pixel usb camera. I set frame rate=60 in aravis tools viewer However it seems that it captures 9-10 frame per second.

**To Reproduce**

**Expected behavior**
I want to take picture at least 30fps by these functions. 
**Camera description:**
 - Manufacturer: Basler
 - Model: acA2040-55uc
 - Interface : USB3

**Platform description:**
 - Aravis version: aravis-0.6.4
 - OS:  Ubuntu 18.04
 - Hardware : arm64 - Jetson Nano

**Additional context**
Its possible to increase buffer size or other parameters to increase image capturing rate? And how to?

EmmanuelP, 2020-01-10:
What are you using for the acquisition, aravis or pypylon ?

barzan-hayati, 2020-01-11:
> What are you using for the acquisition, aravis or pypylon ?

I captured images by `Python` :
[grab.py](https://github.com/basler/pypylon/blob/master/samples/grab.py)
[save_image.py](https://github.com/basler/pypylon/blob/master/samples/save_image.py)

or `C++`, `C` code via Pylon 5.2.0 sample code(SDK):
[pylon 5.2.0 Camera Software Suite Linux ARM 64 bit](https://www.baslerweb.com/en/sales-support/downloads/software-downloads/pylon-5-2-0-linux-arm-64-bit/)
[pylon 5.1.0 Camera Software Suite Linux ARM 64 bit](https://www.baslerweb.com/en/sales-support/downloads/software-downloads/pylon-5-1-0-linux-arm-64-bit/)

I think low value of buffer size has great effect on low frame rate capturing. But via aravis It did not has any latency in live stream video. I want to save captured images to hard (card memory).

I don't know how could I capture images via `C++` or other languages automatically. 

Thanks in advance.

EmmanuelP, 2020-01-13:
I don't understand how this issue is related to aravis. You are using pylon, which is the Basler proprietary API, and has nothing in common with aravis.

Please reopen this issue if you have an issue with aravis.

Cheers.

closed 2020-01-13 by EmmanuelP