AravisProject/aravis #332: Complete arv_gc_feature_node_get_access_mode implementation

madamczyk7ce, 2020-01-10:
As is written in the topic of issue and here:
https://aravis-project.discourse.group/t/loading-genicam-nodes-to-camera/62/7

sh0, 2020-07-04:
@EmmanuelP I am interested in completing `arv_gc_feature_node_get_access_mode` functionality so I did some digging.

As far as I understand, feature nodes should only have `ImposedAccessMode` property and registers should only have `AccessMode` property. Right now feature node implements both `ImposedAccessMode` and `AccessMode`. Is there a reason for this? Maybe some camera manufacturer uses it by mistake?

Current idea is to check referenced nodes using `arv_gc_property_node_get_access_mode` and then impose further restrictions using `ImposedAccessMode` if it is available. So should I also keep `AccessMode` for feature node and check it in addition to `ImposedAccessMode`?

EmmanuelP, 2020-07-05:
Hi Siim,

> @EmmanuelP I am interested in completing arv_gc_feature_node_get_access_mode functionality so I did some digging.

That would be great.

> As far as I understand, feature nodes should only have ImposedAccessMode property and registers should only have AccessMode property. Right now feature node implements both ImposedAccessMode and AccessMode. Is there a reason for this? Maybe some camera manufacturer uses it by mistake?

You are right AccessMode should only be implemented for register nodes and register struct entry nodes.

> Current idea is to check referenced nodes using arv_gc_property_node_get_access_mode and then impose further restrictions using ImposedAccessMode if it is available.

I'm not sure it is the easiest way to handle AccessMode. There could be Converters in the path to the register nodes, so it is not easy to know which registers are affected. I guess it would be better to handle the error reported by the register nodes.

> So should I also keep AccessMode for feature node and check it in addition to ImposedAccessMode?

Outside of register nodes, only ImposedAccessMode should be handled.

EmmanuelP, 2020-08-13:
This is now in master.

closed 2020-08-13 by EmmanuelP