AravisProject/aravis #337: Packet-support problem

madamczyk7ce, 2020-02-10:
As I wrote in topic: https://aravis-project.discourse.group/t/packet-support-problems/72

I use:
aravis 0.6.4
Ubuntu 18.04.4 LTS
Basler acA2040-35gm

EmmanuelP, 2020-04-06:
```
madamczyk
Feb 10

Hi,
I’ve got a problem with packet support, everything is going correctly, but always a first few(1 or 2) frames are invalid. Then I can take 500 000 frames and each of them is received correctly.

    [GvStream::stream_new] Destination stream port = 59071
    [GvStream::stream_new] Source stream port = 49152
    [GvStream::stream_thread] Packet timeout = 40 ms
    [GvStream::stream_thread] Frame retention = 200 ms
    [GvStream::loop] Packet socket method
    [GvStream::set_socket_filter] source ip = 0xac116b10 - port = 49152 - dest ip = 0xac11032d - port 59071
    [GvStream::update_socket] Socket buffer size set to 100000
    [GvStream::find_frame_data] Start frame 2
    [GvStream::missing_packet_check] Maximum number of packet requests reached at dt = 0, n_requests = 113/358
    [GvStream::missing_packet_check] Maximum number of packet requests reached at dt = 0, n_requests = 113/358
    [GvStream::missing_packet_check] Maximum number of packet requests reached at dt = 0, n_requests = 113/358
    [GvStream::missing_packet_check] Maximum number of packet requests reached at dt = 0, n_requests = 113/358
    [GvStream::missing_packet_check] Maximum number of packet requests reached at dt = 0, n_requests = 113/358

In the above log, I used to set fixed-size buffer, but when I am using ARV_GV_STREAM_SOCKET_BUFFER_AUTO nothing is changed, the problem still appears.
I am using the Aravis 0.6.4.
```

EmmanuelP, 2020-04-06:
Which platform are you using ?

Could you try with aravis master ?

EmmanuelP, 2020-08-19:
I'm closing this issue for now. Please reopen if you have the requested informations.

Cheers.

closed 2020-08-19 by EmmanuelP