AravisProject/aravis #338: Black(blank) captured Image: Frame Rate = 0 HZ Incomplete image received, dropping

barzan-hayati, 2020-02-15:
**Describe the bug**

I have a Basler Camera & I want to capture images by aravis viewer.

**To Reproduce**

**Expected behavior**
I could capture images  and view live  stream video using aravis for other Basler camera models. I saw black screen by running avr-viewer . Also by `running arv-device-test` 

```
Looking for the first available camera
vendor        = Basler
model         = daA1600-60uc

(arv-device-test:12802): Aravis-CRITICAL **: 17:53:15.169: arv_gc_string_get_value: assertion 'ARV_IS_GC_STRING (gc_string)' failed
device id     = (null)
sensor width  = 1600
sensor height = 1200
image width   = 1600 (max:1600)
image height  = 1200 (max:1200)
horizontal binning  = 1 (min:1 - max:2)
vertical binning    = 1 (min:1 - max:1)
exposure auto mode  = Continuous
gain                = 64 (min:64 - max:1016)
gain auto mode      = Continuous
trigger selector    = FrameStart
reverse x           = FALSE
payload size  = 3840000 (0x3a9800)
Frame rate = 19 Hz
Frame rate = 20 Hz
Frame rate = 21 Hz
^CProcessed buffers = 0
Failures          = 0
Underruns         = 78
```

` tests/arv-camera-test `

```
Looking for the first available camera
vendor name           = Basler
model name            = daA1600-60uc
device id             = (null)
image width           = 1600
image height          = 1200
horizontal binning    = 1
vertical binning      = 1
payload               = 3840000 bytes
exposure              = 12702 µs
gain                  = 0 dB
uv bandwidth limit     = 163000000 [0..-1]
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Completed buffers = 0
Failures          = 0
Underruns         = 90

```

**Camera description:**
 - Manufacturer: Basler
 - Model: daA1600-60uc (22405944)
 - Interface [e.g. USB3]: USB3

**Platform description:**
 - Aravis version: 0.6
 - OS: Ubuntu 18.04 Bionic  Server
 - Hardware Jetson Nano.

**Additional context**

**How could I resolve this problem while I could see images correctly in Pylon(basler SDK).**

EmmanuelP, 2020-04-06:
Ni @barzan-hayati 

Could you please rerun arv-camera-test with debug output enabled ?

```
arv-camera-test -d all:3
```

l0g1x, 2020-06-10:
Yes it was my cable. CAT5e will not work, only CAT6. Now I'm able to get non-zero frequency for both arv-device-test and arv-camera-test