AravisProject/aravis #340: Interfacing with Testo-885 Thermal Imager

adityabharadwaj14, 2020-03-02:
**Is it possible to interface Testo-885 with the current Aravis Library**
I tried interfacing a Testo-885 Thermal Imager but with no luck. Testo-885 connection was a USB 2.0 with a micro-USB to USB connection. Is it possible to interface with this IR-Camera using the current Aravis Library

**Describe the solution you'd like**
Please let me know if any additional hardware is required or if the Aravis library will work and run it on a LInux environment. If so, then any examples to this would be helpful.

**Describe alternatives you've considered**
I have used the Aravis library to successfully interface with FLIR SC300 series but I also need it to work on Testo-885 Thermal Imager.

dmitryalferov, 2020-03-03:
Probably you are out of luck.
Aravis works with cameras that support GenICam, but as far as my google-fu goes Testo-885 doesn't provide GenICam interface.

EmmanuelP, 2020-04-06:
Thanks @dmitryalferov 

I'm closing this issue, the device has to be Genicam compliant.

closed 2020-04-06 by EmmanuelP