AravisProject/aravis #341: PTGrey/FLIR frame rate setup fails

ArkadiuszRaj, 2020-03-21:
**Describe the bug**
calling **arv_camera_set_frame_rate**(aravis_camera, frame_rate, &error);
fails, error structure:
```
(gdb) p *error
$2 = {domain = 153, code = 1, message = 0x7f700301f0 "node 'AcquisitionFrameRateEnable' not found"}
```
Effectively frame rate is not set.
But  - excerpt from configuration:
```
Enumeration : 'AcquisitionFrameRateAuto'
            EnumEntry   : 'Continuous'
            EnumEntry   : 'Off'
        Boolean     : 'AcquisitionFrameRateEnabled'
        Float       : 'AcquisitionFrameRate'
```
However using arv-tool:
```
$ arv-tool-0.8 control AcquisitionFrameRateEnabled AcquisitionFrameRateAuto AcquisitionFrameRate=31
PointGrey-010AFC78 (USB3)
AcquisitionFrameRateEnabled = true
AcquisitionFrameRateAuto = Off
AcquisitionFrameRate = 31 Hz min:1 max:70.5534
```
frame rate is set
 
**To Reproduce**
Steps to reproduce the behavior:
As described above

**Expected behavior**
A clear and concise description of what you expected to happen.

Frame rate settled? :)

**Camera description:**
 - Manufacturer Point Grey Research  
 - Model Chameleon3 CM3-U3-31S4M
 - Interface USB3
- Firmware FW:1.12.3.00

**Platform description:**
 - Aravis version taken from tag ARAVIS_0_7_3
 - OS: Ubuntu 18.04.4 LTS
 - Hardware ARM64

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2020-03-24:
In the 'GenICam Standard Features Naming Convention`, the recommended feature name is AcquisitionFrameRateEnable, not AcquisitionFrameRateEnabled.

We need to add a special case in ArvCamera.

ArkadiuszRaj, 2020-03-28:
Same case for Point Grey Research Blackfly BFLY-U3-23S6M

```
         Enumeration : 'AcquisitionMode'
            EnumEntry   : 'MultiFrame'
            EnumEntry   : 'SingleFrame'
            EnumEntry   : 'Continuous'
        Command     : 'AcquisitionStart'
        Command     : 'AcquisitionStop'
        Enumeration : 'AcquisitionFrameRateAuto'
            EnumEntry   : 'Continuous'
            EnumEntry   : 'Off'
        Boolean     : 'AcquisitionFrameRateEnabled'
        Float       : 'AcquisitionFrameRate'
        Integer     : 'AcquisitionFrameCount' (Not available)
        Enumeration : 'AcquisitionStatusSelector'
            EnumEntry   : 'FrameTriggerWait'
        Boolean     : 'AcquisitionStatus'
        Enumeration : 'SingleFrameAcquisitionMode'
            EnumEntry   : 'Triggered'
            EnumEntry   : 'FreeRunning'
```

BT in arvcode there is already support for this name
```
case ARV_CAMERA_VENDOR_POINT_GREY_FLIR:
                        arv_camera_set_string (camera, "TriggerSelector", "FrameStart", &local_error);
                        if (local_error == NULL)
                                arv_camera_set_string (camera, "TriggerMode", "Off", &local_error);
                        if (local_error == NULL) {
                                if (priv->has_acquisition_frame_rate_enabled)
                                        arv_camera_set_integer (camera, "AcquisitionFrameRateEnabled", 1, &local_error);
                                else
                                        arv_camera_set_boolean (camera, "AcquisitionFrameRateEnable", TRUE, &local_error);
                        }
                        if (local_error == NULL)
                                arv_camera_set_string (camera, "AcquisitionFrameRateAuto", "Off", &local_error);
                        if (local_error == NULL)
                                arv_camera_set_float (camera, "AcquisitionFrameRate", frame_rate, &local_error);
                        break;
```
I guess that priv->has_acquisition_frame_rate_enabled is not properly set.

ArkadiuszRaj, 2020-03-28:
This fixes my situation, but I am not sure if integer type was mistake or rather needed by some other camera? 

(I am publishing picture for easy identification of needed changes)
[image]

EmmanuelP, 2020-03-28:
Thanks @ArkadiuszRaj 

I have pushed your fix.

Cheers.

closed 2020-03-28 by EmmanuelP