AravisProject/aravis #342: Issue with GST_PLUGIN_PATH

a428tm, 2020-03-25:
**Describe the bug**
Working on implementing AWS's Kinesis Video Stream GStreamer Plugins + Aravis GStreamer + BlackFly GigE camera. Just like Aravis' GStreamer Plugin, AWS requires GST_PLUGIN_PATH to be set. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install AWS Kinesis Video Stream service - https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/gs-send-data.html
2. Install GStreamer Plugin for AWS - https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html
3. Follow the instruction to set path - https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html
4. Once GST_PLUGIN_PATH was set per instructions from both Aravis and AWS, I can only make one work at a time.

**Expected behavior**
It should stream video to AWS console. Tested Aravis setup using Blackfly camera. It streams with no issue locally. Tested using v4l2src (per instruction -https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-gstreamer-plugin.html) and Logitech Webcam for AWS setup. It streamed to console with no issue.

**Camera description:**
 - Manufacturer: FLIR
 - Model: BFLY-PGE-12A2M-CS (https://www.flir.com/products/blackfly-gige?model=BFLY-PGE-12A2M-CS)
 - Interface: GigE

**Platform description:**
 - Aravis version - 0.6.4
 - OS: Ubuntu 18.04
 - Hardware: x86_64

**Additional context**
Based on what I have read, I am trying to use Aravis as GStreamer source and AWS' Kinesis Video Stream as GStreamer sink. I tried setting GST_PLUGIN_PATH for both, but that didn't work. 

This literally is the first (EVER) report/request I am ever making on Github, so if I chose wrong category, i apologize. If I need to provide any more information please do let me know!

Thank you,
Jae

EmmanuelP, 2020-03-25:
Hi Jae,

Could you attach the content of the GST_PLUGIN_PATH variable, and the gst-launch parameters you are using.

Also, you may want to run you pipeline with GST_DEBUG set to 3 or 4, that will output some useful debug informations.

a428tm, 2020-03-25:
Hi Emmanuel,

Thanks for the quick response!

During the installation, I set the variable to `~/.profile` - 
`export GST_PLUGIN_PATH=/usr/local/lib/gstreamer-1.0
`
Then I run this, 
`$ gst-inspect-1.0 aravissrc`

It returned -
> Factory Details:
>   Rank                     none (0)
>   Long-name                Aravis Video Source
>   Klass                    Source/Video
>   Description              Aravis based source
>   Author                   Emmanuel Pacaud <emmanuel@gnome.org>
> 
> Plugin Details:
>   Name                     aravis
>   Description              Aravis Video Source
>   Filename                 /usr/local/lib/gstreamer-1.0/libgstaravis.0.6.so
>   Version                  0.6.4
>   License                  LGPL
>   Source module            aravis
>   Binary package           aravis
>   Origin URL               http://blogs.gnome.org/emmanuel

Then, when I run this 
`$ gst-launch-1.0 aravissrc ! videoconvert ! xvimagesink`
I get the window pop up with the stream. I think that is the correct behavior, correct?

Then, I installed Kinesis Video Stream set GST_PLUGIN_PATH. Then I run -
`$ gst-inspect-1.0 aravissrc`
I get the error that says, "No such element or plugin 'aravissrc'"

I tried running gst-launch-1.0 -
`$ gst-launch-1.0 aravissrc ! videoconvert ! xvimagesink`
Then, i get this error - 

> WARNING: erroneous pipeline: no element "aravissrc"

GST_PLUGIN_PATH returns -
`$ echo $GST_PLUGIN_PATH`
`:/usr/local/lib/gstreamer-1.0:/home/nxlprs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:/home/nxlprs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib`

With that set, when I run -
`$ gst-launch-1.0 v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink stream-name="STREAMNAME" storage-size=512 access-key="ACCESSKEY" secret-key="SECRETKEY" aws-region="REGION" `

This works as expected.

My assumption is that during the installation, Aravis library was installed in `/usr/local/lib/gstreamer-1.0`. I tried copying two files in that into GST_PLUGIN_PATH, but that didn't work (I figured that wouldn't but still tried). I feel like this could be a newbie mistake, but yeah. Any pointer would be appreciated.

Thanks,
Jae

EmmanuelP, 2020-04-06:
You can change the installation path of aravis using ./configure --prefix option.

Nevertheless, you should be able to run your pipeline without changing aravis installation prefix. What is the output of:

```
GST_PLUGIN_PATH=/usr/local/lib/gstreamer-1.0:/home/nxlprs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib:/home/nxlprs/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib GST_DEBUG=4 gst-inspect-1.0 aravissrc
```

EmmanuelP, 2020-08-19:
I'm closing this issue for now. Please reopen if you have the requested informations.

Cheers.

closed 2020-08-19 by EmmanuelP