AravisProject/aravis #345: Black/Glitch in Arv Viewer 0.8

sankhakarfa, 2020-04-20:
I am getting a black or glitch viewer when using arv viewer

When I open the arv-viewer-0.8 from terminal and use the default Fake camera or connect any Machine vision camer

**Platform description:**
 - Aravis version V 0.7.4
 - OS: Ubuntu 18.04
 - Hardware MSI PS63 Modern 8SC

**Additional context**
This happens with aravis version 0.6 also.

[image]

EmmanuelP, 2020-04-21:
Do you use X11 or wayland ?

The video widget based on gtksink gstreamer element is broken. It is slow and does not work reliably on X11.

sankhakarfa, 2020-04-21:
Hi,
It changed to X11 after ubuntu 18.04 upgrade.
Changing it to wayland works fine now.
You can mark it closed now.
Also why arv-viewer is version 0.8 but the release is v0.7.4?

EmmanuelP, 2020-04-21:
I'm keeping this bug open, as we should also support X11.

> Also why arv-viewer is version 0.8 but the release is v0.7.4?

0.8 is the stable series. 0.7.x are unstable development releases, where API may break at any time.

Thanks for the report.

AntonioMichaelAlvarez, 2021-03-19:
Same issue on several machines running Ubuntu 18.04.5 LTS. All using X11 and latest available versions of XFCE or LXDE environments.

EmmanuelP, 2021-03-19:
Works for me on Ubuntu 20.10, using X11.

[image]

I personally don't have the time for debugging the viewer on every distributions. Any help in this area appreciated !

EmmanuelP, 2021-03-19:
I'm closing this bug, as it doesn't seem a generic X11 related issue, but most probably a bug in gtk3 X11 support in Ubuntu 16.04.

closed 2021-03-19 by EmmanuelP

EmmanuelP, 2022-04-10:
https://aravis-project.discourse.group/t/arv-viewer-video-display-problems/89/5

On ubuntu, the issue is fixed by installing gstreamer1.0-gtk3

nznobody, 2022-04-12:
Thanks, this helped me too!