AravisProject/aravis #349: building 0.7.4 on macos 10.15 fails because of missing linux/if_packet.h

avilleret, 2020-04-29:
**Describe the bug**
Build failed with :

    ../src/arvgvstream.c:49:10: fatal error: 'linux/if_packet.h' file not found
    #include <linux/if_packet.h>

**To Reproduce**
Steps to reproduce the behavior:

    brew install gettext intltool gtk-doc libxml2 meson
    brew link --force gettext
    brew link --force libxml2
    meson build
    ninja -C build

**Expected behavior**

I expect the build to not fail.

**Platform description:**
 - Aravis version 0.7.4 (but I do have the same issue with 0.7.3)
 - OS: OSX 10.15.4
 - Hardware x86_64

EmmanuelP, 2020-04-29:
packet-socket is not supported on OSX. I have pushed a fix to master that auto discovers the ability to compile packet-socket.

Meanwhile, you can disable packet-socket using `meson configure -Dpacket-socket=false`

closed 2020-04-29 by EmmanuelP

avilleret, 2020-04-29:
good to know ! thanks for the quick fix !