AravisProject/aravis #350: Starting camera 2nd time (Aravis-Fake-GV01)

cutephoton, 2020-04-30:
**Describe the bug**

A fake camera using `arv-fake-gv-camera-0.8` (`Aravis-Fake-GV01`) fails to stream video 2nd time.

**To Reproduce**

- Launch `arv-viewer-0.8`
- Select `Aravis-Fake-GV01` camera
  - Start video & Stop video
- Select built-in fake video camera
  - Start video & Stop video
- Select `Aravis-Fake-GV01` camera
  - Start video & Stop video

**Expected behavior**

The viewer should start showing the video stream.

Actual results: The UI freezes temporarily when trying to restart `Aravis-Fake-GV01` camera. Video streaming is no longer possible for that camera. I also found the same behaviour in my own application. This issue does not appear to impact the built-in fake camera. (I haven't tested against a real camera.)

Restarting the viewer works fine.

**Camera description:**
Fake

**Platform description:**
 - Aravis version: 0.7.4
 - OS: Fedora 31
 - Hardware: x86_64

cutephoton, 2020-05-03:
So the client was getting stuck in reads waiting for an ack.
https://github.com/AravisProject/aravis/blob/1c56b6d5f25c04aaaa49d7b5d51313496a505fe7/src/arvgvdevice.c#L234

which seems to connect to be connected to the log message here...
https://github.com/AravisProject/aravis/blob/9eeff4224095ed622dc708bd5a62b7def65f71d3/src/arvgvfakecamera.c#L176

Casually reviewing the code, this check would have to fail?
https://github.com/AravisProject/aravis/blob/9eeff4224095ed622dc708bd5a62b7def65f71d3/src/arvgvfakecamera.c#L140

Seems like it might be limited to the fake camera?

The gv fake camera starts emitting the following when streaming is tested. (I need to take a closer look at all this when it's not so late. Or... not... it is a fake camera, after all.)

```
[GvFakeCamera::handle_control_packet] Ignore Write register command 292 (0) not controller
packet_type  = cmd
packet_flags = ack-required
command      = read-register-cmd
size         = 4
id           = 65399
address      =       2560 (0x00000a00)
00000000 42 01 00 80 00 04 ff 77 00 00 0a 00              B......w....    
[GvFakeCamera::handle_control_packet] Read register command 2560 -> 2
packet_type  = ack
command      = read-register-ack
size         = 4
id           = 65399
value        =          2 (0x00000002)
00000000 00 00 00 81 00 04 ff 77 00 00 00 02              .......w....    
[GvFakeCamera::thread] Control packet received
packet_type  = cmd
packet_flags = ack-required
command      = write-register-cmd
size         = 8
id           = 65352
address      =        292 (0x00000124)
value        =          0 (0x00000000)
00000000 42 01 00 82 00 08 ff 48 00 00 01 24 00 00 00 00  B......H...$....

[GvFakeCamera::handle_control_packet] Ignore Write register command 292 (0) not controller
packet_type  = cmd
packet_flags = ack-required
command      = write-register-cmd
size         = 8
id           = 65352
address      =        292 (0x00000124)
value        =          0 (0x00000000)
00000000 42 01 00 82 00 08 ff 48 00 00 01 24 00 00 00 00  B......H...$....

[GvFakeCamera::handle_control_packet] Ignore Write register command 292 (0) not controller
packet_type  = cmd
packet_flags = ack-required
command      = read-register-cmd
size         = 4
id           = 65400
address      =       2560 (0x00000a00)
00000000 42 01 00 80 00 04 ff 78 00 00 0a 00              B......x....    
[GvFakeCamera::handle_control_packet] Read register command 2560 -> 2
packet_type  = ack
command      = read-register-ack
size         = 4
id           = 65400
value        =          2 (0x00000002)
00000000 00 00 00 81 00 04 ff 78 00 00 00 02              .......x....    
```

EmmanuelP, 2020-05-03:
Hi,

I'm not able to reproduce your issue.

Are arv-fake-gv-camera and arv-viewer launched on the same machine ? same account ?
Do you have a non usual network configuration ?

cutephoton, 2020-05-07:
Same machine, same account.I have a bridge & docker (3 IP addresses + local). Do you have thoughts on about network configuration might cause a problem? Maybe I'll try it on another machine if I have time.

EmmanuelP, 2020-05-07:
I think it was fixed by commit 10527f9ad73b248e3aa29ce9fdae757a1309aa93 .

There was a leak of a reference that prevented the proper shutdown of the connection to the camera. Which means at the second attempt, the camera was still used by the dangling refrence.

Could you try aravis master ? Or just this patch ?

cutephoton, 2020-05-09:
That certainly sounds like a likely cause. It certainly makes sense because I observed the remote connection wasn't terminated. I will try this out as soon as I have time. Thanks.

EmmanuelP, 2020-05-10:
I'm closing this report. Please reopen if you are still experiencing this issue with aravis master.

Thanks.

closed 2020-05-10 by EmmanuelP

cutephoton, 2020-05-12:
Yes, this fixed the issue. :) Thanks for the help.